openwrt/luci #7638: luci-proto-bonding:  package missing in 24.10.0

c-spiess, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

After upgrading form 23.05.5 to 24.10.0 the package `luci-proto-bonding` is no longer installed.
Also not shown in package directory.
Bonding itself is running (`proto-bonding` and `kmod-bonding` available).

### Expected behaviour

luci-proto-bonding should be installable

### Steps to reproduce

Upgrade or new install 24.10.0
Configure bonding device

### Additional Information

```shell
{
	"kernel": "6.6.73",
	"hostname": "port-charlotte",
	"system": "AMD GX-412TC SOC",
	"model": "PC Engines apu2",
	"board_name": "pc-engines-apu2",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "24.10.0",
		"revision": "r28427-6df0e3d02a",
		"target": "x86/64",
		"description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
		"builddate": "1738624177"
	}
}
```

### Relevant log output

```shell

```

systemcrash, 2025-02-22:
Bonding package was removed - the standard networking modules can configure it now in 24. All bonding options are available. See #7086

c-spiess, 2025-02-23:
In LUCI i see

[image]
but luci-proto-bonding is not installable.

Bonding is configured in `/etc/config/network` as:
```
config interface 'bond0'
	option proto 'bonding'
	list slaves 'eth0'
	list slaves 'eth1'
	list slaves 'eth2'
	option bonding_policy '802.3ad'
	option min_links '0'
	option ad_actor_sys_prio '65535'
	option ad_select 'stable'
	option lacp_rate 'slow'
	option xmit_hash_policy 'layer2'
	option all_slaves_active '0'
	option link_monitoring 'off'
	option ipaddr '0.0.0.0'
	option netmask '0.0.0.0'

```

The usual suspects `kmod-bonding`, `proto-bonding`, `luci-mod-network` and `netifd` are installed (see [opkg list](https://github.com/user-attachments/files/18930832/opkg.txt))
What am i missing?

c-spiess, 2025-02-24:
Ahh, i got it: the bonding interface isn't needed anymore, instead a bonding device is needed.
So i configured it as:
```
config device
	option type 'bonding'
	option name 'bond0'
	list ports 'eth0'
	list ports 'eth1'
	list ports 'eth2'
	option policy '802.3ad'
	option ipv6 '0'
```
A positive sideeffect is that referencing the device can now be writtten as e.g. `bond0.42` instead of `bond-bond0.42` in 23.05.5.

One further question: is the package `proto-bonding` still needed?
I think this was needed for the bonding `interface`, which is now obsolete.

c-spiess, 2025-02-24:
o.k. then i'll keep it.
Yes, bonding works fine now.

closed 2025-02-24 by c-spiess

bravel03, 2026-11-12:
hola, voy a probar bonding 'balance-rr'