openwrt/luci #7639: luci-mod-network: adding option dnsrr dont generate dnsmasq config

WMP, 2025-02-19:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

I added from LuCI this options:
```
config dnsrr
        option dnsrr 'test.lan.'
        option rrnumber '65'
        option hexdata '74657374'
```

After save and reload i see this options in `/etc/config/dhcp`, but i dont see this options in /var/etc/dnsmasq.conf.dnsmasq1 . Additional, dig dont return me value: test (this is in hex):
```
majanows@EliteBook-840-G9:~$ dig  HTTPS test.lan @192.168.1.1 +short
majanows@EliteBook-840-G9:~$
```

### Expected behaviour

1. I set dns-rr in LuCI
2. this generate option dns-rr in /var/etc/dnsmasq.conf.dnsmasq1
3. dig return me value what i need

### Steps to reproduce

1. Go to DHCP > DNS Records > DNS-RR, set this values:
Resource Record Name: test.lan. (or something other, ex. google.com)
Resource Record Number: 65
Raw Data	Hex Data: test
2. Click to Save & apply

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "Archer-C7-AC1750",
        "system": "Qualcomm Atheros QCA956X ver 1 rev 0",
        "model": "TP-Link Archer C7 v5",
        "board_name": "tplink,archer-c7-v5",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "ath79/generic",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### Relevant log output

```shell

```

djarosz, 2025-03-03:
I am facing same problem. Did bit a little more digging. It looks that startup script at /etc/init.d/dnsmasq looks for 'optiion rrname' not 'option dnsrr' as it is generated by luci. 

After fixing startup script this is options is still not included in generated configuration file but this is ok since startup script uses this to add startup argumet to dnsmasq '--dns-rr=...'. After this Dnsmasq starts responding to queries for tis custom RR record.

But then I have different problem as dnsmasq returns malformed response.  e.g dig my.host HTTPS @192.168.0.1 results in:

;; Warning: Message parser reports malformed message packet.

I am on openwrt 24.10.0 (x86_64)

hnyman, 2025-03-05:
> Did bit a little more digging. It looks that startup script at /etc/init.d/dnsmasq looks for 'optiion rrname' not 'option dnsrr' as it is generated by luci.

DNS-RR has been added by @systemcrash rather recently with f88f903 , so let's cc him about this.

systemcrash, 2025-03-06:
Correct. It should be `rrname`. Fixed in 0a3b60d050526e324c786d98dc53f5a81d45fa6f

closed 2025-03-06 by systemcrash

djarosz, 2025-03-07:
This only resolves first part of my problem. Second is that saving record from UI does not seem to properly encode hexdata filed.

E.g  '1 . alpn="h3,h3"' is encoded as '31202E20616C706E3D2268332C683322' but the valid value should be  '00010000010006026833026832'.  When encoded as  '31202E20616C706E3D2268332C683322' DNS requests result in  Warning: Message parser reports malformed message packet. and no value is received by clients.

The reason is that there should be also rdata class of C_IN encoded as part of rrdata. This script does https://gist.github.com/alxrdn/bc3b2a4beed431d7f1b2f606fe9332e9 generates valid records.

systemcrash, 2025-03-07:
The hex methods are for convenience only. Correct encoding is often dependent on the record type; providing the correct data is an exercise for the user. Maybe some record encoding method could be added. You’re welcome to make a pr with additions.

WMP, 2025-03-08:
Thank you, your fix resolved my previous issue. However, I now have another problem with converting a string to HEX.  

I use **DNS-RR** because I have a **registered domain name** and I use **Cloudflare Tunnels** to expose my home server to the internet. However, when I am on the **same LAN as my server**, I don't want my traffic to go through the internet. So, I have **overridden the A records** on my **OpenWRT**, making the domain resolve to a **local IP address (192.168.1.0/24)**.  

Both locally and globally, the service is accessed over **HTTPS**. The problem started recently – **I can no longer connect to my service from my local network using Android**. I receive the following error:  

```
ERR_ECH_FALLBACK_CERTIFICATE_INVALID
```

After debugging, I found that the issue is caused by the **DNS-RR HTTPS record (TYPE65)**. When I query the global DNS servers for this record, I get:

```
$ dig HTTPS my.domain.com @8.8.8.8 +short
1 . alpn="h3,h2" ipv4hint=198.51.100.10,198.51.100.11 ech=AEX+DQBBrgAgACCd1wd/IPxe8Tsfxs6ZqAmljgxRVKC+nSP8xHznaBaYSQAEAAEAAQASY2xvdWRmbGFyZS1lY2guY29tAAA= ipv6hint=2001:db8::1,2001:db8::2
```

And in **raw HEX format**:

```
$ dig HTTPS my.domain.com @8.8.8.8 +short +unknownformat
\# 136 0001000001000602683302683200040008C633640AC633640B000500 470045FE0D0041AE002000209DD7077F20FC5EF13B1FC6CE99A809A5 8E0C5154A0BE9D23FCC47CE7681698490004000100010012636C6F75 64666C6172652D6563682E636F6D00000006002020010DB800000000 0000000000000001 20010DB8000000000000000000000002
```

---

## **The Problem**
When **Android** connects to my domain, it queries my **local DNS server**, which **forwards the global response without modification**. However, this response **includes public IPv4 and IPv6 hints**.  

As a result, **Chrome detects a mismatch** – it sees that my **A record points to a private IP**, but the **HTTPS record (TYPE65) suggests that the service should be at a public IP**.  
**Because of this, Chrome blocks the connection and shows:**
```
ERR_ECH_FALLBACK_CERTIFICATE_INVALID
```

To fix this, I need my router to **return a different HTTPS record for my domain inside the local network**.

---

## **My Solution**
To resolve this, I extracted **only the beginning of the global DNS response**, keeping only the **`alpn="h2"`** part. I then manually set it in `/etc/config/dhcp` on OpenWRT:

```ini
config dnsrr
        option dnsrr 'test.lan.'
        option rrnumber '65'
        option rrname 'my.domain.com.'
        option hexdata '00010000010003026832'
```

This **works perfectly** and resolves my Android issue:

```
$ dig HTTPS my.domain.com @192.168.1.1 +short
1 . alpn="h2"

$ dig HTTPS my.domain.com @192.168.1.1 +short +unknownformat
\# 10 00010000010003026832
```

---

## **New Issue: How to enter this correctly in LUCI?**
I don't know **what exactly I should enter in the "Raw Data" field in LUCI** to get **the exact same HEX format**. It seems like LUCI's conversion from string to HEX is too simple, and I suspect it's generating incorrect results.

**Can you clarify how I should enter `"alpn=h2"` in LUCI to correctly generate the HEX `00010000010003026832`?**  

---

## **Parsing the HEX Format for TYPE65 (HTTPS) - generated by ChatGPT**
To help clarify how the HEX format works, here’s a breakdown of the correct record (`00010000010003026832`):

| **HEX**        | **Meaning**                                      |
|---------------|------------------------------------------------|
| `00 01`       | Priority `1` (higher = lower priority)         |
| `00 00`       | Alias name (empty, meaning "use the same domain") |
| `01`          | Number of fields (1 field in this record)      |
| `00 03`       | `alpn` parameter (`0x03` means "h2" only)     |
| `02 68 32`    | ALPN value `"h2"` (`h = 0x68`, `2 = 0x32`)    |

To correctly return only **`alpn="h2"`**, the HEX must be:
```
00010000010003026832
```
Any additional data (such as `ipv4hint` or `ech`) can cause issues when used in a local DNS override.