openwrt/luci #7643: luci-proto-vxlan: wrong section type for peers

alaviss, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

The peer is added with `config vxlan_peers`. Potential culprit: https://github.com/openwrt/luci/blob/dbf2c4272a503e9db5ae23b193cb1d62a54b56bc/protocols/luci-proto-vxlan/htdocs/luci-static/resources/protocol/vxlan.js#L182

### Expected behaviour

The peer is added with `config vxlan_peer` (without `s`). This is expected by the vxlan configuration script: https://github.com/openwrt/openwrt/blob/6936deac67e5f0d950bcc26e56bc90b15b12db12/package/network/config/vxlan/files/vxlan.sh#L106

### Steps to reproduce

1. Create a new VXLAN interface
2. Add a peer in the "Peers" tab

### Additional Information

```shell
{
        "kernel": "6.6.77",
        "hostname": "node-01",
        "system": "MediaTek MT7621 ver:1 eco:3",
        "model": "ASUS RT-AX53U",
        "board_name": "asus,rt-ax53u",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "firmware_url": "https://downloads.openwrt.org/",
                "revision": "r28847-e60729c720",
                "target": "ramips/mt7621",
                "description": "OpenWrt SNAPSHOT r28847-e60729c720",
                "builddate": "1739914826"
        }
}
```

### Relevant log output

```shell

```

closed 2025-02-23 by systemcrash

dxdxdt, 2025-07-16:
The typo in vxlan6 has not been patched. Ditto in pr #7646

tkkost, 2025-11-23:
OpenWRT 24.10.4. This error has not been patched yet. The peer is added with 'config vxlan_peers'.

tkkost, 2025-11-23:
> [1f2929f](https://github.com/openwrt/luci/commit/1f2929fb49ada94c8a6aa268df048ce3911ccb4f) picked in July

Hello. I'm a novice and I understand bad what it means. It picked in July but It isn't include to repository yet? And I just should wait, right?

PR #7646 (2025-02-23): luci-proto-vxlan: fix option name: vxlan_peers to vxlan_peer
Fixes: #7643