openwrt/luci #7645: luci wireless overview

olumolu, 2025-02-22:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

Signal icons are not crisp and clear rather extremely hazy. The issue i found out that luci use the signal level as png icons
Like
75% signal
signal-75-100.png
For 50% signal it will be 
signal-50-100.png
And png images are not vector in nature hence if you use a large monitor you will see this images pixelated
And if you add icons larger sized then the total image firmware will be larger

### Expected behaviour

How much you zoom or open in a big sceeen the icons should stay clear.

## solution
Use .svg vector icons which will be even smaller than the png images but due to the nature of vector it can be easily zoomed in or out without pixelated icons

### Steps to reproduce

Network-> wireless

Signal levels

### Additional Information

```shell
Not required
```

### What browsers do you see the problem on?

Firefox, Chrome, Safari, Microsoft Edge

### Relevant log output

```shell

```

olumolu, 2025-02-23:
I can redraw some of them so images can be slowly be replaced.

systemcrash, 2025-02-23:
The iconography hasn't been touched in more than 10 years. It wouldn't be a bad idea to throw this request open to the forums and see if there are some designers who want to see their icons in here for the next decade.

olumolu, 2025-02-23:
Sure that is a good approach. Actially having png as icons in 2025 is not sounds right.

olumolu, 2025-02-23:
https://github.com/openwrt/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/icons/signal-0.png

[image]

olumolu, 2025-02-23:
https://github.com/lessload/openwrt-icon-minimal/tree/main
someone already made this svg we can switch to this new sets of icons

castillofrancodamian, 2025-02-23:
@olumolu In my opinion, monochrome icons are not good.

olumolu, 2025-02-23:
if you add color it will increase the size of the icons i have tested but could be done i dont know how to.

lessload, 2025-03-01:
the original all icon was from https://pictogrammers.com/library/mdi/

lessload, 2025-03-01:
> [@olumolu](https://github.com/olumolu) In my opinion, monochrome icons are not good.

not monochrome svg, file size seem too big for small emmc router

lessload, 2025-03-01:
> So you took icons from the pack? Their license is Apache 2.0 https://pictogrammers.com/docs/general/license/

As i understand, Yes.

lessload, 2025-03-04:
@olumolu error on PR request

systemcrash, 2025-06-29:
Closed by #7747

olumolu, 2025-06-30:
> Closed by [#7747](https://github.com/openwrt/luci/pull/7747)

That pull request did not merged yet why did you close this...?

olumolu, 2025-07-05:
> [@olumolu](https://github.com/olumolu) The PR wasn't merged but pushed directly to the master branch in the [ae5d91d](https://github.com/openwrt/luci/commit/ae5d91da903b6f1d3086d6082ca622231e34f555) So please check the new icons

I just saw that thanks