openwrt/luci #7655: Even if Content-Length exists, luci will add Transfer-Encoding to the response header

refeccd, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

Even if Content-Length exists, luci will add Transfer-Encoding to the response header

### Expected behaviour

Should not contain both Content-Length and Transfer-Encoding

### Steps to reproduce

Using a reverse proxy to proxy openwrt and requesting cgi-bin/cgi-download replicates the issue

### Additional Information

```shell
OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.055.69337~67763a0
```

### Relevant log output

```shell

```

jow-, 2025-02-27:
The issue is with uhttpd, not luci. Probably needs a patch like this:
```diff
diff --git a/proc.c b/proc.c
index 2b69703..89fe320 100644
--- a/proc.c
+++ b/proc.c
@@ -218,6 +218,10 @@ static void proc_handle_header(struct relay *r, const char *name, const char *va
                cl->dispatch.proc.status_code = atoi(buf);
                return;
        }
+       else if (!strcasecmp(name, "Content-Length") ||
+                !strcasecmp(name, "Transfer-Encoding")) {
+               cl->request.disable_chunked = true;
+       }
 
        blobmsg_add_string(&cl->dispatch.proc.hdr, name, val);
 }
```

refeccd, 2025-02-27:
I found an issue much like this one, but no one paid attention to it
https://github.com/openwrt/uhttpd/issues/6

Carseason, 2025-02-27:
I used nginx instead of uhttpd.
uhttpd is not standardized in response or request.
Including uhttpd transfer files, etc.

closed 2025-03-04 by refeccd

webysther, 2025-03-26:
The workaround in nginx:

```nginx
location /cgi-bin/cgi-download {
    proxy_http_version 1.0;
}
```