openwrt/luci #7664: luci-app-xinetd: incorrect handling of REDIRECT option in the UI

lars18th, 2025-03-06:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

Hi,

This is related to commit https://github.com/openwrt/luci/pull/6194

[image]

 

### Actual behaviour

When you want to use the `REDIRECT` option with the `UNLISTED` Type, the `SERVER` option continues to be required. Therefore using the Luci UI you can't add any redirect port. 

### Expected behaviour

The `SERVER` option has to be optional if the `REDIRECT` option is used.

### Steps to reproduce

1. go to: Services -> Xinetd
2. Select "Add new service entry"
3. Type `Servicename` (test), `Port` (11122) and `Redirect` (127.0.0.1:22) and press "Save"

You can't

### Additional Information

```shell
Powered by LuCI openwrt-24.10 branch (25.062.67750~c03591c) / OpenWrt 24.10.0 (r28427-6df0e3d02a)
```

### Relevant log output

```shell

```

systemcrash, 2025-03-17:
@feckert do you have a good idea to fix this?

lars18th, 2025-03-17:
> [@feckert](https://github.com/feckert) do you have a good idea to fix this?

Yes. Here the idea for the fix:

- Convert the "Redirect" box into a select tick (on/off).
- If "Redirect" is off then "Server" box is visible.
- If "Redirect" is on then "Server" box is disabled, and replaced by a "Redirect" box.

What you think?

feckert, 2025-03-19:
@lars18th This should fix your issue https://github.com/openwrt/luci/pull/7689
Can you please report back.

lars18th, 2025-03-19:
> [@lars18th](https://github.com/lars18th) This should fix your issue [#7689](https://github.com/openwrt/luci/pull/7689) Can you please report back.

Hi @feckert ,

Tested and working. Ready to merge. Thank you. 👍

closed 2025-03-19 by feckert

lars18th, 2025-03-19:
Thank you @feckert

lars18th, 2025-03-20:
> Hi [@feckert](https://github.com/feckert) - did you also want to CP these to the 24 branch (the report mentions 24.10.0)?

I've tested it with **22.03** branch too and it works. So it can be backported.

feckert, 2025-03-20:
Old stable (openwrt-23.05):
https://github.com/openwrt/luci/commit/a9654b20f29e757c92d45fc663a6f353822ca66f
https://github.com/openwrt/luci/commit/2e6e4d54d2d6ea223a307f257c481b24bbb919da
Latest stable (openwrt-24.10):
https://github.com/openwrt/luci/commit/36fb21963086b5f86c8c886e5b8737d234c64e44
https://github.com/openwrt/luci/commit/96ad0628528ba4897acb90150745fbf80e8beed9

I have not pushed this to Old Old stable (openwrt-22.03) this is EOL