openwrt/luci #7683: luci-mod-network: link bonding policy not propagated

jpsollie, 2025-03-17:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

when creating a link aggregation  interface with the new netifd system instead of luci-proto-bonding (which was removed a few weeks ago), I enter in luci the option "active-backup" (view screenshot):
, but in /proc/net/bondjp it states round-robin:
```
cat /proc/net/bonding/bondjp 
Ethernet Channel Bonding Driver: v6.6.83

Bonding Mode: load balancing (round-robin)
MII Status: up
MII Polling Interval (ms): 0
Up Delay (ms): 0
Down Delay (ms): 0
Peer Notification Delay (ms): 0

Slave Interface: lan1
MII Status: up
Speed: 1000 Mbps
Duplex: full
Link Failure Count: 0
Permanent HW addr: ba:96:7a:2e:1b:71
Slave queue ID: 0

Slave Interface: sfp-lan
MII Status: up
Speed: 2500 Mbps
Duplex: full
Link Failure Count: 0
Permanent HW addr: ba:96:7a:2e:1b:72
Slave queue ID: 0
root@APBureau4:~# 
```
There's also nothing written in /etc/config/networks about an active-backup policy (though I don't know whether this should be, the documentation isn't clear):

```
config device
        option type 'bonding'
        option name 'bondjp'
        list ports 'lan1'
        list ports 'sfp-lan'
        option primary 'sfp-lan'
        option multicast_router '2'
        option multicast_fast_leave '1'
        option promisc '1'
```
IMO this is a bug, and I hope the missing parameter can be added, but if someone could help me to verify whether uci knows it should be active-backup (which makes it a main openwrt bug), it'd love to hear

### Actual behaviour

a newly created link aggegation interface is started in round-robin mode, whereas it should be as active-backup

### Expected behaviour

the interface is started as active-backup

### Steps to reproduce

[image]

### Additional Information

```shell
{
        "kernel": "6.6.83",
        "hostname": "APBureau4",
        "system": "ARMv8 Processor rev 0",
        "model": "Bananapi BPI-R4",
        "board_name": "bananapi,bpi-r4",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "firmware_url": "https://downloads.openwrt.org/",
                "revision": "r0+29008-0aaabffdea",
                "target": "mediatek/filogic",
                "description": "OpenWrt SNAPSHOT r0+29008-0aaabffdea",
                "builddate": "1742051600"
        }
}
```

### Relevant log output

```shell

```

systemcrash, 2025-03-17:
Any ideas from @hitech95 ?

jpsollie, 2025-03-17:
policy active-backup didn't do what we expected it would do ... do I need to forward this bug to the main repository?

hitech95, 2025-03-17:
> Any ideas from [@hitech95](https://github.com/hitech95) ?

I'll take a look later today.
Thi might be a bug in both luci and netifd.

> policy active-backup didn't do what we expected it would do ... do I need to forward this bug to the main repository?

Can you better explain the expected result so I can later run a test on your config?
What was the configuration with luci-proto-bonding?

jpsollie, 2025-03-18:
the expected result would be:
cat /proc/net/bond/bondjp shows "Bonding Mode: Active-backup"
Attached I'll add my /etc/config/networks with luci-proto-bonding

[network.txt](https://github.com/user-attachments/files/19318609/network.txt)

systemcrash, 2025-03-18:
At a glance it seems the variable simply needs renaming to `bonding_policy`. @jpsollie please try:

```
config device
        option type 'bonding'
        option name 'bondjp'
        list ports 'lan1'
        list ports 'sfp-lan'
        option primary 'sfp-lan'
        option multicast_router '2'
        option multicast_fast_leave '1'
        option promisc '1'
        bonding_policy 'active-backup'
```

jpsollie, 2025-03-18:
> At a glance it seems the variable simply needs renaming to `bonding_policy`. [@jpsollie](https://github.com/jpsollie) please try:
>
> ```
> config device
>         option type 'bonding'
>         option name [...]

nope, didn't work, still round-robin

systemcrash, 2025-03-18:
OK. 

https://github.com/openwrt/netifd/blob/ea01ed41f3212ecbe000422f3c122a01b93fe874/bonding.c#L86-L112 

Source says `policy`. 

Might it be worth testing to see whether the machinery is in place and manually setting bonding itself works with various modes?:
```
ip link add bond0 type bond
ip link set bond0 down
ip link set bond0 type bond mode <mode>
ip link set lan1 down
ip link set lan1 master bond0
ip link set sfp-lan down
ip link set sfp-lan master bond0
ip link set bond0 up
```

hitech95, 2025-03-18:
I confirm tha the node is called 'policy' yesterday I've missed that.

jpsollie, 2025-03-19:
> OK.
>
> https://github.com/openwrt/netifd/blob/ea01ed41f3212ecbe000422f3c122a01b93fe874/bonding.c#L86-L112
>
> Source says `policy`.
>
> Might it be worth testing to see whether the machinery is in place and [...]

ok, I put your commands in a shell script:
```
#!/bin/bash
ip link add bond0 type bond
ip link set bond0 down
ip link set bond0 type bond mode active-backup
ip link set lan1 down
ip link set lan1 master bond0
ip link set sfp-lan down
ip link set sfp-lan master bond0
ip link set bond0 up
```

and executed it, here's what I got:
```
root@APBureau4:~# ip link del bondjp
root@APBureau4:~# sh /tmp/shellscript.sh
root@APBureau4:~# cat /proc/net/b
bnep      bonding/
root@APBureau4:~# cat /proc/net/bonding/bond0
Ethernet Channel Bonding Driver: v6.6.83

Bonding Mode: fault-tolerance (active-backup)
Primary Slave: None
Currently Active Slave: lan1
MII Status: up
MII Polling Interval (ms): 0
Up Delay (ms): 0
Down Delay (ms): 0
Peer Notification Delay (ms): 0

Slave Interface: lan1
MII Status: up
Speed: Unknown
Duplex: Unknown
Link Failure Count: 0
Permanent HW addr: ba:96:7a:2e:1b:71
Slave queue ID: 0

Slave Interface: sfp-lan
MII Status: up
Speed: 2500 Mbps
Duplex: full
Link Failure Count: 0
Permanent HW addr: ba:96:7a:2e:1b:72
Slave queue ID: 0
root@APBureau4:~#
```
so I guess that's what it should be

systemcrash, 2025-03-19:
It looks like the system can do it correctly. Not necessarily netifd. But when you used the proto handler @jpsollie , link bonding also worked?

jpsollie, 2025-03-19:
I assume it did (don't take my word for it): I never had problems with active-backup before, and the policy was set to active-backup on the client PC's networkmanager ...

systemcrash, 2025-04-15:
ping @hitech95

hitech95, 2025-04-15:
> ping [@hitech95](https://github.com/hitech95)

Pong,
if even by specifiing the right key:value in the config/network file it make me think that it might be an issue with netifd.
I can try to take a deeper look this weekend.

systemcrash, 2025-04-15:
Perhaps the necessary bonding setup which [exists here](https://github.com/openwrt/packages/blob/master/net/bonding/files/lib/netifd/proto/bonding.sh) is simply absent.

jpsollie, 2025-04-17:
all right, netifd certainly plays a role here,
this is how I currently see it:
```
config device
        option type 'bonding'
        option name 'bondjp'
        list ports 'lan1'
        list ports 'sfp-lan'
        option multicast_router '2'
        option multicast_fast_leave '1'
        option promisc '1'
        option policy 'active-backup'
        option primary 'sfp-lan'
        option mtu '2000'
        option all_ports_active '1'
```
and /proc/net/bonding/bondjp:
```
cat /proc/net/bonding/bondjp 
Ethernet Channel Bonding Driver: v6.6.85

Bonding Mode: fault-tolerance (active-backup)
Primary Slave: sfp-lan (primary_reselect always)
Currently Active Slave: sfp-lan
MII Status: up
MII Polling Interval (ms): 0
Up Delay (ms): 0
Down Delay (ms): 0
Peer Notification Delay (ms): 0

Slave Interface: lan1
MII Status: up
Speed: Unknown
Duplex: Unknown
Link Failure Count: 0
Permanent HW addr: ba:96:7a:2e:1b:71
Slave queue ID: 0

Slave Interface: sfp-lan
MII Status: up
Speed: 2500 Mbps
Duplex: full
Link Failure Count: 0
Permanent HW addr: ba:96:7a:2e:1b:72
Slave queue ID: 0
```
this is obtained by using this quirk in "local startup":
```
# Put your custom commands here that should be executed once
# the system init finished. By default this file does nothing.

#sometimes sfp-wan needs a restart, no idea why ...
ip link set dev sfp-wan down
ip link set dev sfp-wan up

#sfp-lan is not assigned as primary interface
echo sfp-lan > /sys/class/net/bondjp/bonding/primary
exit 0
```
it certainly is not what I think it should be (eg: i don't think both ports should be active), but it works (for now)

hitech95, 2025-04-20:
> ping [@hitech95](https://github.com/hitech95)

So did some in depth digging on this issue and I've found
 - option `policy` is not set by luci when using active backup, by Netifd dafaults to rr when policy is not set. (already discovered)
 - looks like that the `primary` option is not evaluated. I'm not a Netifd expert.

About the netifd issue I can see that the variable is correctly read from the blobmessage:
 - https://github.com/openwrt/netifd/blob/master/bonding.c#L422 
 - When the `struct bonding_port` is created for each port via `function bonding_create_port`
 - When configuring the port struct is populated the `bool bp->set_primary` is never set.
 - This variable is later used for the call to `function system_bonding_set_port` wich handle the set of the sysfs `bonding/primary`
 - The missing flag also prevent the `function bonding_reset_primary` to properly update the current primary port.

A simple patch like might solve the issue. I'm not a expert of how the call flow of netifd so i'm not sure if this is the right flace to inject such information:
@nbd168 could probably give me some hints about it.

```
diff --git a/bonding.c b/bonding.c
index 3b25751..1d4fc00 100644
--- a/bonding.c
+++ b/bonding.c
@@ -332,6 +332,11 @@ bonding_create_port(struct bonding_device *bdev, const char *name,
        bp->dev.hotplug = hotplug;
        strcpy(bp->name, name);
        bp->dev.dev = dev;
+
+       if (bdev->primary != NULL) {
+               bp->set_primary = strcmp(bdev->primary, name) == 0;
+       }
+
        vlist_add(&bdev->ports, &bp->node, bp->name);
        /*
         * Need to look up the bonding port again as the above
```

hitech95, 2025-04-24:
@systemcrash I've updated the patch above.
That is what i'm testing at the moment.

on a first check it is working. but i'm not sure what was the purpose of `bonding_reset_primary`.
it  _seems_ to be used to properly set the mac address from the primary interface.
it have a sort of fallback that if a primary is not set the first match is taken.
but i'm not sure if this was intended also to be used to automatically set a primary netdev when the primary interface is not configured.

I have indeed double check in the luci side for the fix on the missing `policy` option. My dirty fix (in my test dev) is not properly working and I have to dig deeper in the luci docs.

systemcrash, 2025-04-24:
It looks like you're right. netifd only ever seems to assess the `set_primary` condition, and not assign a value to it at runtime. So your patch looks suitable. Could you put together a PR in the netifd repo? You can add my `Reviewed-by:` tag. 

bonding_reset_primary seems intended to update the MAC address of the bonding device if necessary.

Bear in mind that not all bonds have a primary. Sometimes it's all just slaves, depending on mode (rr, 802.3ad, xor).

systemcrash, 2025-04-24:
Great investigative work BTW. I understand now why my earlier suggestions above didn't work. I forgot the `option` prefix 🤦 

So `        policy 'active-backup'` should have been `        option policy 'active-backup'`

hitech95, 2025-04-24:
Today in IRC nbd found another possible issue.
He provided a patch but I need to test it out.
Unfortunately I didnt had the time today.
I'll take a look tomorrow.

For reference: https://nbd.name/p/040f4b19

systemcrash, 2025-04-25:
Sweet! Good progress :)

hitech95, 2025-05-04:
I have opened a couple of PRs for the issues dicussed above.

systemcrash, 2025-05-17:
openwrt main + 24.10 branch now need to bump netifd to the comit in your netifd PR.

closed 2025-05-17 by systemcrash