openwrt/luci #7685: luci-app-keepalived: Unable to save contents: Permission denied

Jackie264, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

Always shows "Unable to save contents: Permission denied"

### Expected behaviour

Fix it, please.

### Steps to reproduce

1. go to Service --> VRRP --> Notification
2. edit Keepalived.user
3. Save & Apply

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "Velop7C9",
        "system": "ARMv7 Processor rev 5 (v7l)",
        "model": "Linksys WHW03 V2 (Velop)",
        "board_name": "linksys,whw03v2",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "ipq40xx/generic",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### What browsers do you see the problem on?

Safari

### Relevant log output

```shell

```

feckert, 2025-03-19:
I was able to reproduce that. I also saved a new content of `keepalived.user`.
However, a reload of the `/etc/init.d/keepalived` service then also issues the second message that access has been denied.
From my point of view, the `/etc/init.d/keepalived` service does not need to be restarted after changing the `keepalived.user`. I have opened a [PR](https://github.com/openwrt/luci/pull/7688) for this.

@Jackie264 Perhaps you can try this.

closed 2025-03-19 by feckert

feckert, 2025-03-20:
I have also backported this to the current stable branch openwrt-24.10 with https://github.com/openwrt/luci/commit/53679fdfebe28c836bc2dc4ae1b4250169ea999d

PR #7688 (2025-03-19): luci-app-keepalived: do not restart keepalived on user notification change
The content of the file is executed at the next event, so it is not necessary to reload `/etc/init.d/keepalived` if the user notification has changed.

Fixes: #7685 

- [x] This PR is not from my *main* or *master* branch :poop:, but a *separate* branch :white_check_mark:
- [x] Each commit has a valid :black_nib: `Signed-off-by: <my@email.address>` row (via `git commit --signoff`)
- [x] Each commit and PR title has a valid :memo: `<package name>: title` first line subject for packages
- [ ] Incremented :up: any `PKG_VER [... cut]