openwrt/luci #7687: <package-name>: <summary> xiaomi ax3600 luci-app-attendedsysupgrade (24.10.0) error

migueldeblas, 2025-03-18:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

Error in luci-app-attendedsysupgrade
I have a Xiaomi AX3600 IoT
When I try to upgrade (attendedsysupgrade) from 23.05.5. to 24.10.0 Luci does not work. 

The navigation console error:

[image]

### Actual behaviour

When I try to upgrade (attendedsysupgrade) from 23.05.5. to 24.10.0
Luci does not work and does not show any message or error.

### Expected behaviour

Generates the firmware and shows a message or error.

### Steps to reproduce

Luci >  System > Attended Sysupgrade > Search for firmware upgrade > 
Select firmware version: 24.10.0
Board Name / Profile: xiaomi,ax3600
Click "Request firmware image"

### Additional Information

```shell
{
        "kernel": "5.15.167",
        "hostname": "AX3600",
        "system": "ARMv8 Processor rev 4",
        "model": "Xiaomi AX3600",
        "board_name": "xiaomi,ax3600",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05.5",
                "revision": "r24106-10cc5fcd00",
                "target": "ipq807x/generic",
                "description": "OpenWrt 23.05.5 r24106-10cc5fcd00"
        }
}
```

### Relevant log output

```shell

```

systemcrash, 2025-03-18:
Have you updated sysupgrade first?

migueldeblas, 2025-03-19:
Yes, All packages upgraded (via Luci).

systemcrash, 2025-03-19:
@efahl does this look like something you recognise?

efahl, 2025-03-19:
Yes, I'll take a look.

efahl, 2025-03-19:
Displayed error is due to insufficient error handling right here: https://github.com/openwrt/luci/blob/master/applications/luci-app-attendedsysupgrade/htdocs/luci-static/resources/view/attendedsysupgrade/overview.js#L101 .  I'll fix both of those functions to be a bit more robust.
```
$ curl -s https://sysupgrade.openwrt.org/api/v1/revision/24.10.0/ipq807x/generic
{"detail":"Failed to fetch revision for 24.10.0/ipq807x/generic","status":404}

$ curl -s https://sysupgrade.openwrt.org/api/v1/revision/24.10.0/x86/64
{"revision":"r28427-6df0e3d02a"}
```

@migueldeblas The root cause is that the `ipq807x` target has been renamed `qualcommax`, so LuCI ASU will not work for this specific cross-version upgrade on this specific platform.  I suggest you use https://firmware-selector.openwrt.org/?version=24.10.0&target=qualcommax%2Fipq807x&id=xiaomi_ax3600 for this upgrade...  Once you cross the 23.05->24.10 boundary, the ASU clients should work fine.

systemcrash, 2025-03-19:
Seems like target renames should be tracked in general somewhere server-side, as some kind of re-direct?

migueldeblas, 2025-03-19:
If I replace "ipq807x/generic" with "ipq807x/qualcommax" in /etc/ os-release and openwrt_release, as recommended, the request changes: 
https://sysupgrade.openwrt.org/api/v1/revision/24.10.0/ipq807x/qualcommax?1742406082864
but the behaviour is the same:

Uncaught (in promise) TypeError: revision is undefined
    get_revision_count http://192.168.1.1/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-25.070.50735-0322352:2
    applyPackageChanges http://192.168.1.1/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-25.070.50735-0322352:3
    promise callback*applyPackageChanges http://192.168.1.1/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-25.070.50735-0322352:3
    click http://192.168.1.1/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-25.070.50735-0322352:27
    promise callback*handleCheck/</</<.click< http://192.168.1.1/luci-static/resources/view/attendedsysupgrade/overview.js?v=git-25.070.50735-0322352:27
    createHandlerFn http://192.168.1.1/luci-static/resources/ui.js?v=git-25.070.50735-0322352:359

efahl, 2025-03-19:
> "ipq807x/generic" with "ipq807x/qualcommax"

I believe it should be `qualcommax/ipq807x` as in

https://downloads.openwrt.org/releases/24.10.0/targets/qualcommax/ipq807x/

systemcrash, 2025-03-19:
> does this look like a reasonable/proper way to implement the error handling? 
> 
> ```
> 		await Promise.all([
> ...
> 		]);
> ```

Seems reasonable.

closed 2025-03-21 by systemcrash