openwrt/luci #7694: <package-name>:luci-app-aria2 <summary>not work in 24.10 X86_64

PantaFH, 2025-03-21:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

Sat Mar 22 03:36:59 2025 daemon.err aria2c[2109]: Exception caught
Sat Mar 22 03:36:59 2025 daemon.info aria2c[2109]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:54 2025 daemon.info aria2c[2086]: jail: jail (2087) exited with exit: 1
Sat Mar 22 03:36:54 2025 daemon.err aria2c[2086]:
Sat Mar 22 03:36:54 2025 daemon.err aria2c[2086]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:54 2025 daemon.err aria2c[2086]: Exception caught
Sat Mar 22 03:36:54 2025 daemon.info aria2c[2086]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:49 2025 daemon.info aria2c[2015]: jail: jail (2075) exited with exit: 1
Sat Mar 22 03:36:49 2025 daemon.err aria2c[2015]:
Sat Mar 22 03:36:49 2025 daemon.err aria2c[2015]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:49 2025 daemon.err aria2c[2015]: Exception caught
Sat Mar 22 03:36:49 2025 daemon.info aria2c[2015]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:49 2025 daemon.info aria2: Please make sure user 'nobody' has write access to download dir: /mnt/nas/Download/Aria2
Sat Mar 22 03:36:49 2025 daemon.info aria2: Aria2 will run with user 'nobody'.
Sat Mar 22 03:36:18 2025 daemon.info procd: Instance aria2::aria2.main s in a crash loop 6 crashes, 0 seconds since last crash
Sat Mar 22 03:36:18 2025 daemon.info aria2c[1535]: jail: jail (1536) exited with exit: 1
Sat Mar 22 03:36:18 2025 daemon.err aria2c[1535]:
Sat Mar 22 03:36:18 2025 daemon.err aria2c[1535]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:18 2025 daemon.err aria2c[1535]: Exception caught
Sat Mar 22 03:36:18 2025 daemon.info aria2c[1535]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:13 2025 daemon.info aria2c[1523]: jail: jail (1524) exited with exit: 1
Sat Mar 22 03:36:13 2025 daemon.err aria2c[1523]:
Sat Mar 22 03:36:13 2025 daemon.err aria2c[1523]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:13 2025 daemon.err aria2c[1523]: Exception caught
Sat Mar 22 03:36:13 2025 daemon.info aria2c[1523]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:08 2025 daemon.info aria2c[1511]: jail: jail (1512) exited with exit: 1
Sat Mar 22 03:36:08 2025 daemon.err aria2c[1511]:
Sat Mar 22 03:36:08 2025 daemon.err aria2c[1511]: Exception: [Platform.cc:125] errorCode=1 OSSL_PROVIDER_load 'legacy' failed.
Sat Mar 22 03:36:08 2025 daemon.err aria2c[1511]: Exception caught
Sat Mar 22 03:36:08 2025 daemon.info aria2c[1511]: jail: exec-ing /usr/bin/aria2c
Sat Mar 22 03:36:03 2025 daemon.info aria2c[1499]: jail: jail (1500) exited with exit: 1

### Actual behaviour

I installed luci-app-aria on an openwrt 24.10 X86 system, with the same configuration as the previous system, but this error occurred.

### Expected behaviour

It works just as before.

### Steps to reproduce

Use luci to install aria2 normally and write the configuration file,and tried reinstalled, etc.

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "EXWRT",
        "system": "Genuine Intel(R) CPU 0000 @ 2.40GHz",
        "model": "VMware, Inc. VMware20,1",
        "board_name": "vmware-inc-vmware20-1",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "x86/64",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### Relevant log output

```shell

```

hnyman, 2025-03-21:
Likely not related to GUI...

systemcrash, 2025-03-21:
This is not a problem with the luci app, which is just an interface to aria2. Open a bug report in the packages repo where aria2 is.

closed 2025-03-21 by systemcrash