openwrt/luci #7695: <package-name>: <summary>luci-app-rp-pppoe-server: interface field error

migueldeblas, 2025-03-23:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

The Server PPPoE works. I upgrade luci-app-rp-pppoe-server package and Server PPPoE still works but in the luci doesn't see the interface field.

**-- File: /etc/network**

[image]

**-- File: /etc/config/pppoe**

[image]

**-- Luci / Network/Interface (The srvPPPoE interface exists):**

[image]

**-- Luci / Services / RP Service PPPoE Server:**

[image]

**-- I Click in the interface selectBox:**

[image]

**--  If I select @srvPPPoE Interface or lan3.20**  the Server PPPoE doesn't work

[image]

[image]

### Actual behaviour

After upgrade luci-app-rp-pppoe-server from a older version the interface field doesn't show the interface.
In the interface field doesn't show the current interface.
There're two items with srvPPPoE string in the interface selectbox but neither works if i change.

### Expected behaviour

I  expect srvPPPoE to appear as an interface and be selectable.

### Steps to reproduce

After upgrade luci-app-rp-pppoe-server from a older version the interface field doesn't show the interface.
Version with the problem:
luci-app-rp-pppoe-server ->  25.076.76243~9c000ff
rp-pppoe-server ->  4.0-r1

### Additional Information

```shell
luci-app-rp-pppoe-server ->  25.076.76243~9c000ff
rp-pppoe-server ->  4.0-r1

{
        "kernel": "6.6.73",
        "hostname": "AX3600",
        "system": "ARMv8 Processor rev 4",
        "model": "Xiaomi AX3600",
        "board_name": "xiaomi,ax3600",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "qualcommax/ipq807x",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### Relevant log output

```shell

```

systemcrash, 2025-03-24:
Were you already on 24 when you 'upgraded' RP?

migueldeblas, 2025-03-24:
Sorry. I don't remember.  But the package luci-app-rp-pppoe-server in v23 and v24 are the same version.

migueldeblas, 2025-03-26:
I confirmir that @srvPPPoE doesn't work.

systemcrash, 2025-03-26:
And?

systemcrash, 2025-03-26:
> If I change manually the interface to srvPPPoE in the /etc/config/pppoe the service starts and works.

That's what we needed to know.

systemcrash, 2025-03-26:
> Note: Luci changes srvPPPoE to lan3.20

Don't think that's the case: the init script resolves the underlying device (to lan3.20)

systemcrash, 2025-03-26:
Could you try my linked commit above? Download the server file to your device, replacing the existing copy.

jjm2473, 2025-04-30:
@systemcrash Don't try to fix the @ prefix, this problem occurs purely because the init script is not consistent with luci.

https://github.com/openwrt/luci/issues/7755

commit 8838935f (2025-06-10): luci-app-rp-pppoe-server: change interface handler to NetworkSelect

Closes #7695

(cherry picked from commit 1cb4c8b976f6fe180eee46af08b900681934d368)