openwrt/luci #7697: luci-app-banip: Support for HTTP 1.1

webysther, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

Related to #7432 

If enable websockets support or add support for HTTP 1.1 breaks banip UI.

### Expected behaviour

Support HTTP > 1.0

### Steps to reproduce

1. Fresh install openwrt
2. Install a reverse proxy: apache, nginx and etc
3. Enable support for HTTP 1.1

### Additional Information

```shell
{
"kernel": "6.6.73",
"hostname": "HIDE",
"system": "12th Gen Intel(R) Core(TM) i5-12600H",
"model": "QEMU Standard PC (i440FX + PIIX, 1996)",
"board_name": "qemu-standard-pc-i440fx-piix-1996",
"rootfs_type": "ext4",
"release": {
"distribution": "OpenWrt",
"version": "24.10.0",
"revision": "r28427-6df0e3d02a",
"target": "x86/64",
"description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
"builddate": "1738624177"
}
}
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell

```

systemcrash, 2025-03-24:
Only the banIP UI? @dibdot

webysther, 2025-03-24:
> Only the banIP UI? [@dibdot](https://github.com/dibdot)

Yeap, I found the tty require websocket to work (#7022), but I don't use.

dibdot, 2025-03-24:
banIP is using LuCI standards and it works in usual setups. For unknown reason this seems not to work ... please fix your setup.

```
return view.extend({
	load: function () {
		return Promise.all([
			L.resolveDefault(fs.read_direct('/etc/banip/banip.custom.feeds'), ''),
			L.resolveDefault(fs.read_direct('/etc/banip/banip.feeds'), ''),
			L.resolveDefault(fs.read_direct('/etc/banip/banip.countries'), ''),
			uci.load('banip')
		]);
	},
```

webysther, 2025-03-25:
> banIP is using LuCI standards and it works in usual setups. For unknown reason this seems not to work ... please fix your setup.
>
> ```
> return view.extend({
> 	load: function () {
> 		return [...]

Hi, sorry I don't get it, how I need to fix my setup?

systemcrash, 2025-03-25:
custom feeds, feeds and countries?

dibdot, 2025-03-26:
> how I need to fix my setup?

I have no idea, I'm using Openwrt/LuCI standard (uhttpd without any proxy) and it just works, e.g.:

[image]

To get the culprit in your setup I would update banip & luci-app-banip to the latest version (your screenshot is outdated), restart the rpcd daemon and than open a developer console and check the network communication.

webysther, 2025-03-26:
> I have no idea, I'm using Openwrt/LuCI standard (uhttpd without any proxy) and it just works, e.g.:
>
> [image]
>
> To get the culprit in your setup I would update banip & luci-app-banip to the latest [...]

I did this, update all packages, restart the router and **only the banip not working** in the related https://github.com/openwrt/luci/issues/7432 other users have the same problem, access direct works, this don't mean that will works in another situation like in reverse proxy.

To give you more data, only two requests broke:

[image]

The first download works because the response is empty (I guess).

[image]

This two broke this two dropbox:

[image]

The behavior to force downgrade to HTTP 1.0 maybe is a sign about another problem.

Another think I notice, the country selection works but the list is incomplete which means the block with this 3 requests is buggy somehow:

[image]

To give more context, this proxy run 87 hosts for my homelab, only the tab of banip ip have this problem.

webysther, 2025-03-26:
Another think, when running cgi-exec works, but this tab is the only one that uses cgi-download, maybe changing to cgi-exe if is not complex fixes the problem but not what causes it.

webysther, 2025-03-26:
Edit Allowlist also broke:

[image]

The blocklist is empty and the request works, looks like only broke if has data in the cgi-download.

webysther, 2025-03-26:
I confirmed that I change only for `/cgi-bin/cgi-download` will work:

```nginx
location /cgi-bin/cgi-download {
    proxy_http_version 1.0;
}
```

webysther, 2025-03-26:
Maybe related to https://github.com/openwrt/packages/issues/10134 https://github.com/openwrt/luci/issues/3140

webysther, 2025-03-26:
And seems not justify to use cgi-download:

https://github.com/openwrt/luci/blob/9c94c7dc93195e43b5102dfb0adf93e08d31c52c/modules/luci-base/htdocs/luci-static/resources/fs.js#L335-L343

dibdot, 2025-03-26:
> And seems not justify to use cgi-download:

Why not? The information is not relevant for ubus.

jow-, 2025-03-26:
This is likely the same as https://github.com/openwrt/luci/issues/7655, basically an uhttpd issue.

closed 2025-03-26 by webysther