openwrt/luci #7698: Cannot read properties of null (reading 'wifi')

siwee, 2025-03-24:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

Hi,

I downloaded the latest firmware version 24.10.0 from https://downloads.openwrt.org/releases/24.10.0/targets/x86/64/. When logging into the OpenWrt main page, I encountered the error message "Cannot read properties of null (reading 'wifi')". However, refreshing the page made the error disappear. 

It’s worth mentioning in advance that the host running OpenWrt only has a single wired network interface and no wireless WiFi device.

[image]

### Actual behaviour

When logging into the OpenWrt main page, I encountered the error message "Cannot read properties of null (reading 'wifi')". 

### Expected behaviour

My device only has one wired network card and no wireless WiFi equipment, but this error message shouldn’t be appearing anyway!

### Steps to reproduce

This error message doesn’t appear every time, and I’m unable to determine the steps to consistently reproduce it.

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "OpenWrt",
        "system": "AMD Ryzen 9 7950X 16-Core Processor",
        "model": "VMware, Inc. VMware Virtual Platform",
        "board_name": "vmware-inc-vmware-virtual-platform",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "x86/64",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell
luci.js?v=25.014.55016~7046a1c:186 Uncaught (in promise) TypeError: Cannot read properties of null (reading 'wifi')
    at Object.hasSystemFeature (luci.js?v=25.014.55016~7046a1c:186:81)
    at initNetworkState (network.js?v=25.014.55016~7046a1c:55:373)
    at ClassConstructor.getDSLModemType (network.js?v=25.014.55016~7046a1c:209:217)
    at ClassConstructor.load (interfaces.js?v=25.014.55016~7046a1c:36:458)
    at ClassConstructor.__init__ (luci.js?v=25.014.55016~7046a1c:139:219)
    at ClassConstructor.super (luci.js?v=25.014.55016~7046a1c:15:42)
    at new ClassConstructor (luci.js?v=25.014.55016~7046a1c:5:11)
    at luci.js?v=25.014.55016~7046a1c:175:135
```

siwee, 2025-03-24:
> This is a known 'feature' for devices that do not have Wi-fi directly accessible in their hardware. Luci scans for wifi assuming it is present, doesn't find it, and just flags a warning. Do you see the warning multiple times over time or just once after login?

Hi, @systemcrash 

Yes, I’ve seen it multiple times. It appears suddenly, but I can’t figure out the pattern behind its occurrence. Additionally, for OpenWrt, is this prompt really necessary? I think users of OpenWrt have certain expectations about their devices, and this prompt might make them feel that the firmware has a bug or an incorrect configuration.

systemcrash, 2025-03-25:
> Hi, [@systemcrash](https://github.com/systemcrash)
>
> Yes, I’ve seen it multiple times. It appears suddenly, but I can’t figure out the pattern behind its occurrence. Additionally, for OpenWrt, is [...]

One may be able to dress this up as a general notice about lack of wifi by handling this case specifically. 

I recall there was a bit of code merged for 24 in the (underlying) openwrt system about not assuming wifi is present (e.g. for VMs), but there's no 'machinery' in place for luci to access this yet.

foxhui, 2025-05-06:
I also have this problem, but I don't know how it is triggered. Switching to each interface may trigger this prompt and seriously affect the use (because it will make the settings of this page blank). Only restarting the system can alleviate this problem, but it may appear again the next time you use it. Below is my hardware information and system version, as well as the link where the failure will occur. Although it contains third-party plug-ins, this plug-in has no problem in version 23.05.5.

Device：FriendlyElec NanoPi R2S
System Version：OpenWrt 24.10.1 r28597-0425664679 / LuCI openwrt-24.10 branch 25.123.45881~e061a5f

[image]

[image]

FriendlyGremlin, 2025-05-07:
I'm facing the same issue. In my case, it happens when clicking on Network --> Interfaces. If I refresh the page, the error disappears. It seems to appear only after a fresh login. Version is `OpenWrt 24.10.1 r28597-0425664679 / LuCI (HEAD detached at 2ac26e56) branch 25.103.51521~2ac26e5`, Model `GL.iNet GL-MT6000` with Architecture: `ARMv8 Processor rev 4`.

Error message in Developer console reads:

```
luci.js?v=25.103.51521~2ac26e5:186 Uncaught (in promise) TypeError: Cannot read properties of null (reading 'wifi')
    at Object.hasSystemFeature (luci.js?v=25.103.51521~2ac26e5:186:81)
    at initNetworkState (network.js?v=25.103.51521~2ac26e5:55:373)
    at ClassConstructor.getDSLModemType (network.js?v=25.103.51521~2ac26e5:209:217)
    at ClassConstructor.load (interfaces.js?v=25.103.51521~2ac26e5:36:458)
    at ClassConstructor.__init__ (luci.js?v=25.103.51521~2ac26e5:139:219)
    at ClassConstructor.super (luci.js?v=25.103.51521~2ac26e5:15:42)
    at new ClassConstructor (luci.js?v=25.103.51521~2ac26e5:5:11)
    at luci.js?v=25.103.51521~2ac26e5:175:135
```

hunghvu, 2025-12-29:
> Potential fix in [db91dbc](https://github.com/openwrt/luci/commit/db91dbcdf7db16b307aaf054a6039a6abc464b0c) - can you take it for a spin [@hunghvu](https://github.com/hunghvu) ?

@systemcrash I cannot reproduce the bug anymore. Previously the bug consistently showed up when opening *Interface* as a new tab.

My device is x86 without Wi-fi. I updated Luci to version *25.360.65088~78a62f2* which includes your commit, so I think the bug is resolved, at least for my use case. Thanks!