openwrt/luci #7700: <luci-proto-wireguard	>: <Cannot generate user configuration>

venzislav, 2025-03-25:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

I installed openwrt on new router and add wireguard server as usual and then I add client and when I click on "generate configuration" button on the bottom nothing happens and cannot generate user configuration :

[image]

And I don't know if this is connected to the issue but the qrencode command that generate QR give me this error when I test it from console :

root@OpenWrt:~# qrencode -o test.png dasdasdas
PNG output is disabled at compile time. No output generated.

I hope to be fast fix because I have clients waiting :(

### Expected behaviour

To generate user configuration 

### Steps to reproduce

go to Network > interface > click on edit in the wireguard interface > Peers > User Edit > Configuration Export
Generate Configuration 

### Additional Information

```shell
{
	"kernel": "5.15.167",
	"hostname": "OpenWrt",
	"system": "ARMv8 Processor rev 4",
	"model": "ASUS TUF-AX4200",
	"board_name": "asus,tuf-ax4200",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "23.05.5",
		"revision": "r24106-10cc5fcd00",
		"target": "mediatek/filogic",
		"description": "OpenWrt 23.05.5 r24106-10cc5fcd00"
	}
}
```

### Relevant log output

```shell

```

systemcrash, 2025-03-25:
Check that the luci uqr lib is installed. The updated WG proto uses that now. Check your browser console for (JS) errors logged.

venzislav, 2025-03-25:
> Check that the luci uqr lib is installed. The updated WG proto uses that now. Check your browser console for (JS) errors logged.

I have the uqr .
Here the errors in browser console but it does not mean nothing for me :( : 

[192.168.1.1-1742923074150.log](https://github.com/user-attachments/files/19453104/192.168.1.1-1742923074150.log)

venzislav, 2025-03-25:
I also tried to uninstall all wireguard modules and reinstall them and the same problem , also I did few different interfaces and the same problem :(

systemcrash, 2025-03-25:
> `wireguard.js?v=git-25.080.48774-47f75f5:53 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'indexOf')`

There are only two instances of `indexOf` in the code - did you configure IPv6?

venzislav, 2025-03-26:
> > `wireguard.js?v=git-25.080.48774-47f75f5:53 Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'indexOf')`
> 
> There are only two instances of `indexOf` in the code - did you configure IPv6?

No - I don't use v6 :

[image]

systemcrash, 2025-03-26:
Did you fill in the system hostname?

systemcrash, 2025-03-26:
Not that most people can reach an RFC1918 IP but you just shared a secret key with the world. I recommend you delete that message from the edit history @venzislav ...

venzislav, 2025-03-26:
> Not that most people can reach an RFC1918 IP but you just shared a secret key with the world. I recommend you delete that message from the edit history [@venzislav](https://github.com/venzislav) ...

no no - it is just a test to see if it will fix the problem :) and no secret - I will do the all stuff again 
Thanks for the note !

systemcrash, 2025-03-26:
Should be resolved by 4207c2c5d51470fd273441f92b0a2c104441454c