openwrt/luci #7705: Feature request: Allow setting rootfs size in luci-app-attended-sysupgrade

0ldMonk, 2025-03-29:
### What would you like to see in luci?

Currently, sysupgrade image builder supports setting rootfs size. Even owut command has -S option to set larger rootfs size like 1 GB. It would be great if luci-app-attended-sysupgrade also had configuration to set rootfs size of upgrade image. 

This is the only reason i have to login via ssh and use owut to upgrade instead of using luci-app-attended-sysupgrade gui.

aparcar, 2025-03-29:
Sounds like a good feature and it should be complicated, please feel free to create a PR

efahl, 2025-03-29:
This has been on my list for a long time, but I've held off because:

1) It's sort of problematic in that only certain platforms should allow resizing.  I've been trying to figure out how to restrict the option in `owut` to only valid platforms, but haven't been able to determine a generic way to do that.

2) Every time you change root fs size, sysupgrade wipes your config (https://github.com/efahl/owut/issues/27) so you need to restore from backup manually.

systemcrash, 2025-03-29:
@efahl I've been thinking about a couple of homogeneous functions in openwrt recently and this falls squarely into that class. 

This one likely a big json/shell script, with a list of those platforms that do support it, growing manually through user contributions. At worst, a big override checkbox so users can play with the sharp knives and inevitably cut themselves.

At least in the config case, stashing the config in a tmpfs/ramfs seems like a way forward. Altho that already seems to be present in sysupgrade:

```sh
...
	[ "$(rootfs_type)" = "tmpfs" ] && {
		echo "Cannot save config while running from ramdisk." >&2
		ask_bool 0 "Abort" && exit
		return 0
	}
```

efahl, 2025-06-07:
I had forgotten about this issue and just stumbled across it again...

The feature has been add to snapshot, with a PR pending for inclusion in 24.10.  All the caveats mentioned above still hold.  Setting the value requires using CLI as it is an Extremely Advanced Dangerous Feature™.  Read https://openwrt.org/docs/guide-user/installation/sysupgrade.owut#expanding_root_file_system completely before using it.

https://github.com/openwrt/luci/commit/42d32dff7f718d27ae7ccc65ebc83d001ebf14e3

systemcrash, 2025-06-07:
So can this be closed? It's in master and 24.10 already.

closed 2025-06-08 by systemcrash