openwrt/luci #7707: luci-mod-network: `CTRL+F` search results include elements that are not currently visible

vincejv, 2025-03-31:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

Searching for "Bridge device" on Interfaces tab shows 7/7 results even if there is no "Bridge device" shown visibly in that tab
[image]

"Bridge device" is only visible in devices tab
[image]

### Actual behaviour

The browser says it finds the search term even though it is currently not visible in the current tab, as shown in the screenshot

### Expected behaviour

Search shouldn't return any results when searching for _term_ that's not currently visible

### Steps to reproduce

1. Go to Luci > Network > Interface
2. Press `CTRL+F` to bring up search in browser
3. Type any search term that applies to the bug reported

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "Pictor",
        "system": "Intel(R) Core(TM) i7-2600 CPU @ 3.40GHz",
        "model": "Dell Inc. OptiPlex 790",
        "board_name": "dell-inc-optiplex-790",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "x86/64",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell
n/a
```

systemcrash, 2025-03-31:
Visibility and presence are two different things: search finds entities present in the page. 'tabs' are just a paradigm of stacking information, which CSS and JS can modify display behaviour, depending on how 'tabs' are structured (separate JS pages or all pages in one JS).

Given that putting each tab into a separate JS page is a performance and file-system (later down the image compile chain: compression) compromise, all tabs in one file is more efficient, page-load wise. 

One conceivable way to change this behaviour is to high-jack the search to a custom function, limited in scope to what is indexed. Otherwise this is currently default browser behaviour.

vincejv, 2025-03-31:
Yes I'd like to make search function like how facebook and other normal sites do it , search should only include visibile elements and not just because it's present in the DOM. Example, when message box is hidden in facebook, the contents of the message box is not included in the search even if it's in DOM, example here is i searched for the word "visibility"

[image]

[image]

vincejv, 2025-04-02:
also to add, it seems like this bug can only be reproduced (as far as I have tried) in `luci-mod-network` not in other luci apps like `luci-app-package-manager` so idk why `luci-mod-network` would be any special?

SimplyProgrammer, 2025-09-19:
I would argue that this on its own, is a somewhat unsolvable problem as it seems to be heavily dependent on a theme you are using (among the other things).
For example, when I am using the current default theme "Bootstrap", also this proves that it can be reproduced outside of the Network tab (I suppsoe that is what you mean by luci-mod-network), since I am in System > System > Language and Style, and I search for "System", I get 11 results (N/11) while I can actually only see 3...

[image]

But for instance, when I switch to Argon and do the same, this problem disappears almost completely in this particular case:

[image]
(N/3 = correct)<br>

As was already stated, probably the only real solution that would work the same across all of the themes and possibly also browsers (as I have shown in your other issue, browser can and likely does metter) is to simply not rely on a loosely defined behavior of a certain browser (such as Chrome, that is on top of that susseptable to unpredictable changes) but to embed or, as was said "highjack", the browser search (Ctrl+F) and make a custom OpenWrt/Luci content specific searching functionality for it. Essentially, make it the feature of Luci web interface itself, which essentially brings us pretty much to what I have suggested in my [Search bar for Luci #7918](https://github.com/openwrt/luci/issues/7918)...