openwrt/luci #7708: luci-app-firewall: `CTRL+F` search does not return search results correctly on some browsers

vincejv, 2025-04-02:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

Says it can't find the search term as per the screenshot, works correctly on Firefox however..

### Expected behaviour

Should say 1/1 search since the search term is in the web page visibile

### Steps to reproduce

1. Go to Luci > Network Firewall > Traffic Rules
2. Press `CTRL+F` to bring up browser search functionality
3. Enter search term that can be found in the page
4. Does not return anything as reported by browser as seen in the screenshot

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "Pictor",
        "system": "Intel(R) Core(TM) i7-2600 CPU @ 3.40GHz",
        "model": "Dell Inc. OptiPlex 790",
        "board_name": "dell-inc-optiplex-790",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "x86/64",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell

```

SimplyProgrammer, 2025-09-17:
I can tell you that this happens because the leftmost column of each table row (tr) is, for whatever reason (likely as an extreme/excessive form of optimization) rendered as a styled `::before` contents of which seems to be ignored in a browser search...

SimplyProgrammer, 2025-09-19:
Also, sorry for discrediting your issue. I have problems with searching for stuff as much as you have...<br> However, I have to further argue that this should not have been reported as being an issue with Luci nor OpenWrt as the Ctrl+F feature is not part of the Luci web interface, nor was the Luci web interface designed to be completely compatible with this (as its exact functionality differs heavily from browser to browser).
This is arguably the problem of Chrome's rather clumsy implementation of the feature than anything OpenWrt related, so If you want, you should report this particular thing to Google...

To illustrate my point, I have decided to try this with Firefox (which is often regarded as superior in polish), and indeed it works as expected:

[image]
(So if you are really desperate, consider ^ to be a possible solution but...)
This is also why I am advocating for in-luci search capable of searching for OpenWrt-related features and contents so we would not have to rely on loosely goosy, browser-specific behaviors, sometimes buggy Ctrl+F behaviors and have a dedicated thing for the job instead, which is something I think you also [agree with](https://github.com/openwrt/luci/issues/7918#issuecomment-3300751989)...

vincejv, 2026-01-07:
> problem of Chrome's rather clumsy implementation of the feature rather than anything OpenWrt related

@SimplyProgrammer quick update, it doesn't work on safari too