openwrt/luci #7711: Feature request: expose LDPC (low density parity check) in wireless menu

professor-jonny, 2025-04-03:
### What would you like to see in luci?

I'm wondering if the LDRP setting can be exposed in the wireless devices settings page under luci?

With the Xbox 360 console it does not connect to a wireless network unless set to legacy (802.11abg) or have this setting turned off in a 802.11N or above network.
Practically all the internal wifi modules in Xbox 360 slim basically won't connect to any modern network. for all devices that have a marvell wireless chipset.
The issue presents it's self as device is not connected to a network when it is.

The effected wifi modules devices 1501 and 1400 which are almost exclusively fitted to all slim type consoles.
One of the two variants for the phat console (WKS168) is also affected.
That is a potential for a good chunk of 84 million units as shipped worldwide to be impacted.

disabling the setting enables normal use with an 802.11n or above network:
uci set wireless.radio.1.ldpc='0'
uci commit wireless.

https://techinfodepot.shoutwiki.com/wiki/Microsoft

systemcrash, 2025-04-04:
Doable.

systemcrash, 2025-04-05:
What does `iw list` output when you have `uci set wireless.radio.1.ldpc='0'`? Do you also have ` uci set wireless.radio.1.rxldpc='0'`?

closed 2025-04-07 by systemcrash

systemcrash, 2025-04-08:
Once I get a few more fixes ironed out, I can CP this to 24/23.

systemcrash, 2025-04-08:
CP'd to 24 via daf09c2473d208f994456d65b78176489e35184f