openwrt/luci #7715: luci-theme-bootstrap: Network > Firewall, incorrect zones width

janeblower, 2025-04-09:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

When you enable forwarding from/to any zone the width of div .zone-dest breaks, it becomes greater than 40% > breaks the behavior of the entire div .zone-forwards.

### Expected behaviour

[image]
I want the div width to be as it was before (as shown in the screenshot I found on google).

### Steps to reproduce

1. go to Network > Firewall
2. edit any zone
3. allow forward _from/to_

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "OpenWrt",
        "system": "Intel(R) Core(TM) i5-6200U CPU @ 2.30GHz",
        "model": "QEMU Standard PC (i440FX + PIIX, 1996)",
        "board_name": "qemu-standard-pc-i440fx-piix-1996",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "x86/64",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### Relevant log output

```shell

```

systemcrash, 2025-04-11:
This behaviour is intentional. Perhaps @feckert might have a suggestion. BTW @feckert perhaps the string 'all others' can be made i18n able :)

janeblower, 2025-04-11:
> This behaviour is intentional. Perhaps [@feckert](https://github.com/feckert) might have a suggestion. BTW [@feckert](https://github.com/feckert) perhaps the string 'all others' can be made i18n able :)

we ca try to change _**flex-wrap**_
before:
```css
.zone-forwards {
    display: flex;
    flex-wrap: wrap;
}
```
[image]

after:
```css
.zone-forwards {
    display: flex;
    flex-wrap: nowrap;
}
```

[image]

systemcrash, 2025-04-12:
This is... acceptable. Although you must remember to treat all of the other themes equally.

janeblower, 2025-04-14:
@systemcrash 
I fixed all the themes.

closed 2025-04-15 by systemcrash