openwrt/luci #7716: luci-app-adblock: cosmetic bug 'Blocked Domains' count

itzViking, 2025-04-10:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

But logs show

Thu Apr 10 20:38:34 2025 user.info adblock-4.2.7-r3[2707]: blocklist with overall **107 928** blocked domains loaded successfully (Raspberry Pi 3 Model B Plus Rev 1.3, bcm27xx/bcm2710, OpenWrt 24.10.0 r28427-6df0e3d02a )

### Actual behaviour

I ran `opkg upgrade adblock luci-app-adblock` and the cosmetic bug appeared. Wasn't happening with the version that was released around the same time as OpenWRT 24.10.0

### Expected behaviour

luci UI should be consistent with the logs, in my case it should show 107 928 instead of 107.

107928 could also work, or with any other thousands separator.

### Steps to reproduce

Add any blocklist that has a count over 3 digits to reproduce?
Haven't confirmed, but seems like the space separating the thousands is wrongly parsed. 

### Additional Information

```shell
Model	Raspberry Pi 3 Model B Plus Rev 1.3
Architecture	ARMv8 Processor rev 4
Target Platform	bcm27xx/bcm2710
Firmware Version	OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.037.68331~42f464c
Kernel Version	6.6.73
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell
Thu Apr 10 20:38:34 2025 user.info adblock-4.2.7-r3[2707]: blocklist with overall 107 928 blocked domains loaded successfully (Raspberry Pi 3 Model B Plus Rev 1.3, bcm27xx/bcm2710, OpenWrt 24.10.0 r28427-6df0e3d02a )
```

systemcrash, 2025-04-10:
ping @stangri

itzViking, 2025-04-11:
This is probably fixed in the last version, I ran opk upgrade luci-app-adblock again, and after a reboot, the luci UI is now consistent with the logs.

closed 2025-04-11 by dibdot