openwrt/luci #7720: luci-proto-wireguard: Untitled peer in status view after checkPeerHost() implementation

elfman03, 2025-04-11:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

For VPN configurations that specify EndPoint Host there is a lingering issue with matching up the peer name when checkPeerHost is run in rpcd/ucode/luci.wireguard which causes "Untitled Peer" to be shown.  In particular, if a wireguard peer has a volatile dns result

[image]

[image]

### Actual behaviour

"Unknown peer" is shown for the peer in the wireguard status

/overlay/upper/usr/share/rpcd/ucode/luci.wireguard has a new addition (#7342, #7347, #7453)

It looks like checkPeerHost logic was added to support having multiple peers that use the same key.  Instead of simply mapping the name based on the key as was done before commit 7acea81 It is implemented with an additional code block that resolves the endpoint_host into an ip address via popen(resolveip) and comparing that to the active ip address of the endpoint.

Unfortunately, many VPN providers use volatile dns address mappings for their endpoints so invoking resolveip will return different IP addresses each time which may or may not include the active peer IP address that was locked in when the interface was started.

For example, I use surfshark with endpoint host "us-dal.prod.surfshark.com"  When the interface starts is will map to an a lookup of that host (e.g., 2.56.189.80).    However, when the luci status page is viewed, it runs resolveip again which may not include the address that was from when the wireguard interface was started.  For example, running resolveip ten times back to back for the host reports.  Rarely will the active peer ip address show up in the dns mapping.

root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
212.102.40.83
212.102.40.76
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
212.102.40.83
212.102.40.76
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
212.102.40.76
212.102.40.83
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
2.56.189.90
2.56.189.82
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
2.56.189.82
2.56.189.90
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
2.56.189.90
2.56.189.82
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
2.56.189.82
2.56.189.90
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
2.56.189.90
2.56.189.82
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
2.56.189.82
2.56.189.90
root@fairview-master:/overlay/upper/usr/share/rpcd/ucode# resolveip us-dal.prod.surfshark.com
2.56.189.90
2.56.189.82

Rarely will the the IP address that the endpoint is connected to show up in the dynamic results of resolveip for this type of VPN endpoint and thus the endpoint will show up as "Untitled peer" in the status page except for a few seconds now and then when the resolveip circles back in the server dns rotation to the ip address that is active on the interface.

The easiest way to fix would be to roll back the code commits for #7343 and related bugs.  

Another option might be to somehow save the peer name-> address mapping that was done at the time the interface was brought up and look that up instead of trying to dynamically use resolveip against the name every time the status view is updated.

### Expected behaviour

The name configured in the wireguard peer configuration should be shown.

### Steps to reproduce

1. create a wireguard interface with a explicit endpoint host (e.g., 
config wireguard_surfshark
        option description 'Surfshark - Dallas'
        option public_key '<<REDACT>>'
        list allowed_ips '0.0.0.0/0'
        option endpoint_host 'us-dal.prod.surfshark.com'
        option persistent_keepalive '25'
        option endpoint_port '51820'

2. start the interface
3. view status in luci (Status->Wireguard)

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "fairview-master",
        "system": "ARMv7 Processor rev 5 (v7l)",
        "model": "Google WiFi (Gale)",
        "board_name": "google,wifi",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "ipq40xx/chromium",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell

```

systemcrash, 2025-04-12:
I noticed this too but it didn't bother me too much - perhaps you have a good idea @this-username-has-been-taken ?

systemcrash, 2025-04-12:
I think falling back to the peer key as a best effort is a solution, especially if the peer is still 'untitled'.

this-username-has-been-taken, 2025-04-13:
Hello!

Thank you for tagging me! There is no way we can get history of IPs which were proved by the DNS service other than writing them to some file (or remembering in any other way). But it is an overkill for the case.
I agree with @systemcrash - let's fallback to an old algorithm if the new one fails.

I suggest the following solution.
Change
```
uci.foreach('network', `wireguard_${last_device}`, (s) => {
	if (!s.disabled && s.public_key == record[1] && (!s.endpoint_host || checkPeerHost(s.endpoint_host, s.endpoint_port, record[3])))
		peer_name = s.description;
});
```
to
```
uci.foreach('network', `wireguard_${last_device}`, (s) => {
	if (!s.disabled && s.public_key == record[1] && (!s.endpoint_host || checkPeerHost(s.endpoint_host, s.endpoint_port, record[3])))
		peer_name = s.description;
});

if (!peer_name) {
	uci.foreach('network', `wireguard_${last_device}`, (s) => {
		if (s.public_key == record[1])
			peer_name = s.description;
	});
}
```

If this solution is OK - I will make a PR.

Here is an updated `luci.wireguard` file: [luci.wireguard.zip](https://github.com/user-attachments/files/19724143/luci.wireguard.zip)

@elfman03 , can you please check if it fixes the issue for you? To do this:
1. Back up original file `/overlay/upper/usr/share/rpcd/ucode/luci.wireguard`
2. Replace the file with the provided one.
3. Reboot the router.
4. Wait for the peer's IP address to change and check if peer name is defined.

systemcrash, 2025-04-13:
Then one might as well just do something like this so we don't loop again:
```diff
						let peer_name;
+						let _pn;

						uci.foreach('network', `wireguard_${last_device}`, (s) => {
+							if (s.public_key == record[1])
+								_pn = s.description;
							if (!s.disabled && s.public_key == record[1] && (!s.endpoint_host || checkPeerHost(s.endpoint_host, s.endpoint_port, record[3])))
								peer_name = s.description;
						});

+						if (!peer_name) peer_name = _pn;
```
The point of the exhaustive search is because you have multiple peers with the same key, one possibly disabled, or not connected.

this-username-has-been-taken, 2025-04-13:
Agree!
I have modified variable name to make it more clear.
The PR is here: https://github.com/openwrt/luci/pull/7722
I have also tested in on a live OpenWrt system.

systemcrash, 2025-04-13:
Closed by 8bff54dc47057e36c0ceb263efe4c82580003924

closed 2025-04-13 by systemcrash

elfman03, 2025-04-13:
Wow, y'all are fast!  I verified this patch on my system too.  Works great!.  Thanks.