openwrt/luci #7729: <package-name>: <summary> (TP-Link Archer C60 v2) After 24.10.x update, router does not turn up (does not start) 5.8 GHz radio.

Undergroundbr, 2025-04-16:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

After update 24.10.x, router does not start (does not start) 5.8 GHz radio. Used as a repeater (5.8in/2.4out). 5.8 GHz radio starts normally with firmware version 23.05.x.

BAD: openwrt-24.10.1-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade

### Actual behaviour

CURRENT:
Hostname: Generic
Model: TP-Link Archer C60 v2

### Expected behaviour

Architecture: Qualcomm Atheros QCA956X ver 1 rev 0
Target Platform: ath79/generic
Firmware Version: openwrt-24.10.1-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade

### Steps to reproduce

BAD: openwrt-24.10.1-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade

### Additional Information

```shell
openwrt-24.10.1-ath79-generic-tplink_archer-c60-v2-squashfs-sysupgrade
```

### Relevant log output

```shell

```

hnyman, 2025-04-16:
Bug might be real, but not related to LuCI gui.

kofec, 2025-04-16:
https://github.com/openwrt/openwrt/issues/17175
https://github.com/openwrt/openwrt/issues/14541