openwrt/luci #7732: Can not save network selection for wireless device

cardboardaardvark, 2025-04-18:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

In the "General Setup" tab of the "Interface Configuration" for "Wireless Network: Access Point" after selecting a network clicking the save button does not do anything. Clicking the save button works when changing other things.

The workaround is to edit /etc/config/wireless by hand then issue wifi reload

### Expected behaviour

The save button should close the modal dialogue and save the configuration.

### Steps to reproduce

1) Navigate to Network -> Wireless from the menu at the top of the luci web page
2) Click edit for the network associated with a radio
3) Click the General Setup tab in the bottom half of the page
4) Click the drop down box next to "Network"
5) Check the box for the vlan I want the wireless network to use
6) Click the save button

I am able to use the save button if I remove the check mark from the vlan I want to use. Attempting to put it back results in clicking the save button not doing anything again.

### Additional Information

```shell
Model: Netgear R6260
Target platform: ramips/mt7621
Firmware version: OpenWRT 24.10.1 r28597-0425664679 / LuCI (HEAD detached at 2ac26e56) branch 25.103.51521~2ac26e5
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell

```

systemcrash, 2025-04-18:
Do you get any kind of traceback in the browser console in this case? Tools -> Browser Tools -> Console

cardboardaardvark, 2025-04-18:
> Do you get any kind of traceback in the browser console in this case? Tools -> Browser Tools -> Console

No, there is no output in the javascript console related to this. The only output on the console is a warning about password fields present on a page not using TLS. 

I just ran through the reproduction steps starting with unchecking the vlan associated with the wireless network. I noticed that there is not a significant delay in UI responsiveness when unchecking the vlan and there is about a 1/4 or 1/2 second delay when checking the vlan. The delay here being specifically when clicking away from the drop down with check boxes inside it.

> how are your VLANs configured

VLANS were configured entirely using luci

> what kind of VLAN is it?

Are you asking if this hardware uses DSA? It uses DSA. If that's not the question please clarify.

tobiaswaldvogel, 2025-04-18:
> Maybe this is related to your recent addition [@tobiaswaldvogel](https://github.com/tobiaswaldvogel) ?
> 
I don`t think that it is related to this in any way. My change was related to displaying wifi vlans, but as far as I understand this is just about attaching a wifi network to a particular ethernet vlan.
So I guess the setup is just as usual: You are trying to select an interface, which is defined as a bridge connected to your vlan adapter. I tried to reproduce that but in my case that works perfectly well.

Could you share your /etc/config/network ?

cardboardaardvark, 2025-04-18:
> as far as I understand this is just about attaching a wifi network to a particular ethernet vlan

correct

> You are trying to select an interface, which is defined as a bridge connected to your vlan adapter.

Also correct. Specifically WiFi is on vlan 8 and normal network clients are on vlan 7 and administration happens on vlan19. This access point sits as a leaf so it only has an IP address for administration purposes, bridges wifi to vlan 8, and rips off the tags to expose vlan 7 on two of the switch ports. The wan interface is present but is unused.

> Could you share your /etc/config/network ?

Sure. I'll attach it with this response.

[network config.txt](https://github.com/user-attachments/files/19815739/network.config.txt)

cardboardaardvark, 2025-04-19:
Here's some more context. When I was setting up the wireless device initially I fully configured it, including selecting the vlan to bridge with, and was unable to use the save button. For lack of any other idea I clicked dismiss and went back into the config and saw my configuration was still there and clicked save and that worked. What I missed was that the vlan selection was not included in the config that was still present.

srbouffard, 2025-04-22:
I've been having the same issues. Unable to define Wireless networks on a fresh install of openwrt. The Save button does nothing at all. No consol, popup or network outputs with errors.

jrmclaugh, 2025-04-22:
Also having the issue on 24.10.1 fresh install to an MT6000.

Wifi settings all save fine if I just don't attach to a VLAN.  Save button does nothing once I do.  No errors in browser console, tested on Firefox and Chrome on Fedora 42 and Kubuntu 24.04.2

Something in the debugging process allowed me to save one interface attached to a VLAN yesterday after initially seeing this error, not sure what the difference is on this next interface (same VLAN, just different wifi interface).  The same is true across all VLANs I try to attach.

Also used opkg upgrade to update all luci-related packages available, same issue.

srbouffard, 2025-04-22:
Was able to narrow it down to the same issue where trying to assign a wireless SSID to a VLAN-backed interface wouldn’t save in LuCI. After some digging, I found a reliable workaround:

Instead of attaching the SSID directly to the VLAN-backed interface, wrap it in a dedicated bridge device and attach the wireless to that:

1. Go to Network → Interfaces → Devices
2. Add a new bridge device (wrapper) 
    - Set bridge port to the target VLAN interface
3. Then go to Network → Interfaces
    - Edit your VLAN-backed interface
    - Change its device to the wrapper
4. Now in Network → Wireless, you’ll be able to assign the SSID to the "bridge wrapper" and it will save correctly

Worked great for me on an R7800. Hope it helps others running into the same!

jrmclaugh, 2025-04-22:
I was able to workaround just using uci commands:
uci show | grep ssid
uci set wireless.<appropriate wifinetN for desired SSID from grep>.network=<VLAN interface name>
uci commit

Changes reflected in luci and able to use the Save button again changing any other wifi-related settings.  It seems just committing the wifi network to a VLAN has some issue.

Kveri, 2025-04-22:
I can confirm the same issue on my side as well. After I select Device to be assigned to Wireless the Save button does nothing. If I choose a device and instead of clicking Save I click Disable button at the top of Wireless config it gives this error: "Assignment to constant variable."

systemcrash, 2025-04-23:
>  If I choose a device and instead of clicking Save I click Disable button at the top of Wireless config it gives this error: "Assignment to constant variable."

Can you share the traceback from the browser console for that?

MakaanPL, 2025-04-24:
Same issue with Mi R3G (also MT7621). On 24.10 I can attach WiFi SSID to specific interface (with br-lan.XX device) normally, on 24.10.1 I cannot. "Save" button seems inactive. It works only with "lan" selected as the only option, I can remove it and add again.
**This seems to be just LuCi issue.**

systemcrash, 2025-04-24:
A fix went out already, so it will be in the next luci-mod-network available when the packages are rebuilt.

damadmai, 2025-04-25:
I assume this fix is meant: https://github.com/openwrt/luci/commit/1e3aa83b449b17fb3a09441d1a6c837752718c59
Thanks!

systemcrash, 2025-04-25:
It's live now. Do an owut or sysupgrade.

cardboardaardvark, 2025-04-25:
> It's live now.

Awesome! Thank you.

> Do an owut or sysupgrade.

I'd love to. Except. owut will not install via opkg and luci-app-attendedsysupgrade isn't a package that opkg is aware of at all.

owut fails to install with " * opkg_install_pkg: Checksum or size mismatch for package owut. Either the opkg or the package index are corrupt. Try 'opkg update'." but I did the opkg update using the one-liner from the OpenWRT docs:

```
root@garage:~# opkg update && opkg install owut
Downloading https://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/kmods/6.6.86-1-2a6d68f3ded425efc0850296e7f62d53/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/kmods/6.6.86-1-2a6d68f3ded425efc0850296e7f62d53/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/telephony/Packages.sig
Signature check failed.
Remove wrong Signature file.
Installing owut (2025.04.08~ef2bfb4d-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/packages/owut_2025.04.08~ef2bfb4d-r1_all.ipk
Collected errors:
 * opkg_install_pkg: Checksum or size mismatch for package owut. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package owut.
root@garage:~# 
```

For luci-app-attendedsysupgrade: opkg just does not know about it, though the package system is operational and is up to date as I just updated it trying to install owut:

```
root@garage:~# opkg info luci-app-attendedsysupgrade 
root@garage:~# opkg info owut
Package: owut
Version: 2025.04.08~ef2bfb4d-r1
Depends: libc, attendedsysupgrade-common, rpcd-mod-file, ucode, ucode-mod-fs, ucode-mod-ubus, ucode-mod-uci, ucode-mod-uclient, ucode-mod-uloop
Status: unknown ok not-installed
Section: utils
Architecture: all
Size: 27930
Filename: owut_2025.04.08~ef2bfb4d-r1_all.ipk
Description: The OpenWrt Upgrade Tool (owut) is a command line program that
 checks for new packages, package build breakages, generates
 sysupgrade images containing installed packages and installs
 images.  It is written in ucode for easy end-user customization.
 
 Documentation is available at
 https://openwrt.org/docs/guide-user/installation/sysupgrade.owut

root@garage:~# 
```

But I am able to install attendedsysupgrade-common:

```
root@garage:~# opkg install attendedsysupgrade-common
Installing attendedsysupgrade-common (9) to root...
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/packages/attendedsysupgrade-common_9_all.ipk
Installing rpcd-mod-rpcsys (2024.09.17~9f4b86e7-r1) to root...
Downloading https://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/base/rpcd-mod-rpcsys_2024.09.17~9f4b86e7-r1_mipsel_24kc.ipk
Configuring rpcd-mod-rpcsys.
Configuring attendedsysupgrade-common.
root@garage:~# 

```

If I use luci itself to try to manage the package install it is also unaware of luci-app-attendedsysupgrade even after updating opkg from the command line and from luci itself. Screen shot attached.

Wondering if I could do the upgrade entirely with packages I also looked at the updates tab in luci with the contents "filter" being empty and there are no entries in the list of packages that can be updated.

[image]

17LbCat, 2025-04-27:
> > It's live now.
> 
> Awesome! Thank you.
> 
> > Do an owut or sysupgrade.
> 
> I'd love to. Except. owut will not install via opkg and luci-app-attendedsysupgrade isn't a package that opkg is aware of at all.

You can just go to system -> software -> click update lists, grab the new version of luci-mod-network.

I can confirm that solved it for me. I also updated the other luci packages.

closed 2025-04-27 by systemcrash

mloska, 2025-06-16:
I could- i installed newest 25 version and I can observe same issue. However I was able to make the workardound suggested earlier to create a "bridge wrapper" for vlan and then set the WiFi to the interface connected to the "wrapper bridge" instead of lan