openwrt/luci #7737: LuCI (HEAD detached at 2ac26e56)

4nth5, 2025-04-22:
### How can we otherwise help?

What is the reason for this signature? Does this affect stability?
TP-Link Archer C6U v1
Firmware Version: OpenWrt 24.10.1 r28597-0425664679 / LuCI (HEAD detached at 2ac26e56) branch 25.103.51521~2ac26e5
Upgrade from 24.10.0 to 24.10.1

hnyman, 2025-04-22:
My answer from your thread in forum:

>  As far as I understand, the official firmware 24.10.1 for Archer C6U is not in the main OpenWrt repository branch

Actually, none of the releases are in the main branch, because they are always in separate release branches. There is one release branch for 24.10, and the 24.10.1 is directly on that branch: https://git.openwrt.org/?p=openwrt/openwrt.git;a=shortlog;h=refs/heads/openwrt-24.10

[image]

But, I think that if you git checkout a tag and build from that, you are not any more "on branch", but git thinks that you are specifically at a certain commit (referenced by the tag) independent of any branch, so you are at "detached" status.

LuCI has no fixed release tags:     https://git.openwrt.org/?p=project/luci.git;a=shortlog;h=refs/heads/openwrt-24.10

Instead, the tagging on OpenWrt main repo sets also the feeds of the release commit to point to a specific commit.
Like https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=d183d7bb7827a469f09bf77f2f22fd9d70ac0ed6;hp=04256646797da82a3a0c419d298ca7719192d128

`+src-git luci https://git.openwrt.org/project/luci.git^2ac26e56cc55102cb10e7b0867c2b78e0f6d5fd8`

So, if you build 24.10.1 tag, you get LuCI as of commit 2ac26e56cc.

(no comment about the unrelated topic of "not very successful upgrade")

Ps. In general I find it better to build not from exact release tag, but the HEAD of that release branch (like openwrt-24.10), so that you get also the possible later fixes for that branch.

subnut, 2025-04-22:
Ultimately, what's shown on there is decided by this piece of code -
https://github.com/openwrt/luci/blob/d40c03ad8dfdc22259fbafaeec70a8f60e324e11/luci.mk#L119-L131

subnut, 2025-04-22:
Okay, I have _finally_ gotten to the root of this issue.  
The way luci is cloned is different between the two releases...

### v24.10.0
```sh
git clone 'https://git.openwrt.org/project/luci.git' './feeds/luci'
cd './feeds/luci'
git checkout -b '7b0663a5557118499dc3b3d44550efc1b6fa3feb' '7b0663a5557118499dc3b3d44550efc1b6fa3feb'
cd -
```

### v24.10.1
```sh
git clone --depth 1 'https://git.openwrt.org/project/luci.git' './feeds/luci'
cd './feeds/luci'
git fetch --depth=1 origin '2ac26e56cc55102cb10e7b0867c2b78e0f6d5fd8'
git -c advice.detachedHead=false checkout '2ac26e56cc55102cb10e7b0867c2b78e0f6d5fd8'
cd -
```

4nth5, 2025-04-22:
@subnut So does the problem really exist? I thought we could close the topic.

hnyman, 2025-04-22:
> > Firmware Version: OpenWrt 24.10.1 r28597-0425664679 / LuCI (HEAD detached at https://github.com/openwrt/luci/commit/2ac26e56cc55102cb10e7b0867c2b78e0f6d5fd8) branch 25.103.51521~2ac26e5
> > Upgrade from 24.10.0 to 24.10.1

> The change was introduced by this commit - https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462

@subnut 
Thanks for digging that change up.

So, the commit https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462 from OpenWrt PR https://github.com/openwrt/openwrt/pull/18003 merged by @robimarko  in March (and backported to 24.10) changed the checkout style of feeds to "shallow". And 24.10.1 was the first release done after that change. So the change became visible with that release. 

This is only cosmetics, but still the string `LuCI (HEAD detached at https://github.com/openwrt/luci/commit/2ac26e56cc55102cb10e7b0867c2b78e0f6d5fd8) branch 25.103.51521~2ac26e5` with the "detached" may look strange. It looks "detached" because due to the shallow (single-commit) checkout, there is no commit history and the commit can't be placed into a branch.

There is probably not much cure for that, as with an one-commit shallow history for the build, the commit will remain detached. That versioning string maybe looks strange with official releases which are built from a locked single commit of each feed. 
 
cc @jow-

robimarko, 2025-04-22:
Sadly, it's an unexpected side effect of the linked commit.

That being said, its cosmetics only, since the goal of the commit is to save on bandwidth and time.

subnut, 2025-04-22:
@robimarko Maybe we could hard-code the value of PKG_GITBRANCH upon tagging a new release?

robimarko, 2025-04-22:
I am not sure if its really worth the man-time to do so since its purely cosmetical

hnyman, 2025-04-22:
One simple option might be to patch luci.mk in the release branch after it has been created, to always return the correct release branch name.

Thanks to the `?=` used there the simplest change would just be one-line set action in the start of luci.mk

hnyman, 2025-04-22:
Two alternative one-line solutions:

1) Patch luci.mk. Just add a hard-coded definition before the conditional setting of the variable.

```
--- a/luci.mk
+++ b/luci.mk
@@ -116,6 +116,8 @@ PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_NAME)
 PKG_PO_VERSION?=$(if $(DUMP),x,$(strip $(call findrev)))
 PKG_SRC_VERSION?=$(if $(DUMP),x,$(strip $(call findrev,1)))
 
+PKG_GITBRANCH:=LuCI test openwrt-24.10 test branch
+
 PKG_GITBRANCH?=$(if $(DUMP),x,$(strip $(shell \
        variant="LuCI"; \
        if git log -1 >/dev/null 2>/dev/null; then \
```

2) Patch luci-base Makefile instead, as the version.uc belongs to that, and no other LuCI package in the repo refers to that variable.

```
--- a/modules/luci-base/Makefile
+++ b/modules/luci-base/Makefile
@@ -9,6 +9,8 @@ include $(TOPDIR)/rules.mk
 PKG_NAME:=luci-base
 PKG_BUILD_FLAGS:=no-lto
 
+PKG_GITBRANCH:=LuCI another test openwrt-24.10 branch
+
 LUCI_TYPE:=mod
 LUCI_BASENAME:=base
```
 
Based on tests on build-system, both approaches work:

```
perus@ub2504:...neon-vfpv4_musl_eabi$ cat root-ipq806x/usr/share/ucode/luci/version.uc 
export const revision = '25.108.47076~f7b7478', branch = 'LuCI openwrt-24.10 branch';

perus@ub2504:...neon-vfpv4_musl_eabi$ cat root-ipq806x/usr/share/ucode/luci/version.uc 
export const revision = '25.108.47076~f7b7478', branch = 'LuCI test openwrt-24.10 test branch';

perus@ub2504:...neon-vfpv4_musl_eabi$ cat root-ipq806x/usr/share/ucode/luci/version.uc 
export const revision = '25.108.47076~f7b7478', branch = 'LuCI another test openwrt-24.10 branch';
```

Applying one of these alternatives would not help 24.10.1 builds, but all following 24.10.x release builds should have the correct branch name.

(I haven't yet tested these with a real tag / locked feed commit, but I see no reason why they would not work)

hnyman, 2025-04-23:
See PR #7745

4nth5, 2025-04-24:
@hnyman  Good news. Thank you.

PR #7745 (2025-04-26): [24.10] luci-base: Hardcode 'openwrt-24.10' as the branch name
Hardcode 'openwrt-24.10' release branch as the branch name to avoid `LuCI (HEAD detached at 2ac26e56) branch` being shown as the branch.

Feeds are since March 2025 cloned shallow if cloned from a specific commit, like done with a release tag. Change in OpenWrt: https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462

As the only cloned commit of a release tag is without history, the branch can't be identified.

So, hardcode the branch name to get the proper branch name `LuCI openwrt-24.10 branch` into ve [... cut]
commit cca46543 (2025-04-26): luci-base: Hardcode 'openwrt-24.10' as the branch name

Hardcode 'openwrt-24.10' release branch as the branch name to avoid
'LuCI (HEAD detached at 2ac26e56) branch' being shown as the branch.

Feeds are since March 2025 cloned shallow if cloned from a specific
commit, like done with a release tag. Change in OpenWrt:
https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462

As the only cloned commit of a release tag is without history, the branch
can't be identified.

So, hardcode the branch name to get the proper branch name
'LuCI openwrt-24.10 branch'.

Fixes #7737
commit ca53a4a8 (2025-04-26): luci-base: Hardcode 'openwrt-23.05' as the branch name

Hardcode 'openwrt-23.05' release branch as the branch name to avoid
'LuCI (HEAD detached at 2ac26e56) branch' being shown as the branch.

Feeds are since March 2025 cloned shallow if cloned from a specific
commit, like done with a release tag. Change in OpenWrt:
https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462

As the only cloned commit of a release tag is without history, the branch
can't be identified.

So, hardcode the branch name to get the proper branch name
'LuCI openwrt-23.05 branch'.

Fixes #7737
 [... cut]
commit 149d68ed (2025-04-26): luci-base: Hardcode 'openwrt-23.05' as the branch name, fixed

Fix the previous commit, really set the 23.05 branding, not 24.10.

Hardcode 'openwrt-23.05' release branch as the branch name to avoid
'LuCI (HEAD detached at 2ac26e56) branch' being shown as the branch.

Feeds are since March 2025 cloned shallow if cloned from a specific
commit, like done with a release tag. Change in OpenWrt:
https://github.com/openwrt/openwrt/commit/32d0a57dc10811a24816e51f10e55963f40fe462

As the only cloned commit of a release tag is without history, the branch
can't be identified.

So, hardcode the branch na [... cut]