openwrt/luci #7740: Feature request: Tilt/Pan for Dlink camera DCS-5030L

aguaviva, 2025-04-22:
### What would you like to see in luci?

I successfully got pan&tilt  working using this micropython script:
``` python
import os  # Import os to run shell commands

# coil sequence to move stepper motors
vertical = [14, 15, 16, 17]
horizontal = [18, 19, 20, 21]

def move(iterations, vel, commands):
    p = 0
    for i in range(iterations):
        str1 = "gpioset -c 0 -t0 %i=0" % commands[ p % len(commands)]
        str2 = "gpioset -c 0 --hold-period 2ms -t0 %i=1" % commands[ (p+vel) % len(commands)]
        os.system(str1+";"+str2)
        p+=vel

    str1 = "gpioset -c 0 -t0 %i=0" % commands[ p % len(commands)] #deenergize stepper coil
    os.system(str1)

def test():
    move(300, -1, horizontal)
    move(100, 1, vertical)
    move(300, 1, horizontal)
    move(100, -1, vertical)

if __name__ == "__main__":
    test()
```
However I failed miserably writing a lucy module (too complex), any help on how to create a lua module that show 4 buttons or how to add them to mjpg-streamer is very welcome.

systemcrash, 2025-04-23:
I recommend writing the above in ucode, which runs on the router, and calls the named commands. Check through the repo for .uc files. Then the JS front end can call the ucode script. 

For guidance, some good examples to check are luci-app-ddns, and luci-proto-wireguard.

aguaviva, 2025-04-23:
What is the JS frontend? I'd hope a lot having some clear documentation and a sample. I'd love to contribute to this fantastic project to be honest.

aguaviva, 2025-04-26:
Before that I need to do a few things, I just sent a PR (my first PR!) to expose the LEDs, SD card and buttons of my device. My next step is to give a name to the GPIOs that drive the stepper motors but it didn't work very well, anyway I am not sure if this is the right approach. 

Is this something you can help with?