openwrt/luci #7751: luci-i18n-opkg-lang : all translate package missing in 24.10?

vndroid, 2025-04-28:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

Why? How to solve?

### Expected behaviour

Is it because it is about to be abandoned and switched to apk?

### Steps to reproduce

no i18n pkg.

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "OpenWrt",
        "system": " Intel",
        "model": "VMware, Inc."
        "board_name": "wmware-inc-vmware20-1",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0",
                "revision": "r28427-6df0e3d02a",
                "target": "x86_64",
                "description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
                "builddate": "1738624177"
        }
}
```

### Relevant log output

```shell

```

hnyman, 2025-04-28:
There is no luci-app-opkg in 24.10, as it has been replaced with luci-app-package-manager (which has translations quite normally).

closed 2025-04-28 by hnyman

vndroid, 2025-04-28:
Yeah, it is right!