openwrt/luci #7754: luci-app-firewall: There is an unknown symbol at the end of the word "Clone"

dvlom99, 2025-04-29:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

When viewing the firewall rules, the Сlone button caption has a symbol that my browser does not display correctly.

### Expected behaviour

I suggest changing the button caption to simply "Clone".

### Steps to reproduce

Go to Luci > Network Firewall > Traffic Rules

### Additional Information

```shell
{
        "kernel": "6.6.86",
        "hostname": "ax53u",
        "system": "ARMv8 Processor rev 4",
        "model": "Cudy WR3000S v1",
        "board_name": "cudy,wr3000s-v1",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.1",
                "revision": "r28597-0425664679",
                "target": "mediatek/filogic",
                "description": "OpenWrt 24.10.1 r28597-0425664679",
                "builddate": "1744562312"
        }
}
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell

```

hnyman, 2025-04-29:
That is probably a too fancy unicode symbol, like discussed in  #7746 , originating from https://github.com/openwrt/luci/commit/22cccf7b042e537b1e1b99fdae8bf18fa646b997 by @systemcrash 

It seems to cause problems for several users so far. 
Yes, it is an unicode symbol, but quite unnecessary on that button.
Might be wise to remove it.

@jow- 
Do we have stance toward this kind of symbols?

systemcrash, 2025-04-29:
This happens when users set custom fonts or a font with limited Unicode coverage in their browser. The symbol is from Unicode 3.0 - 15 years old and is available in the majority of fonts.

dwfreed, 2025-05-01:
I'm not sure where "is available in the majority of fonts" comes from.  This is a screenshot from https://fonts.google.com/noto/fonts?preview.text=%E2%BF%BB&preview.layout=grid which is just the Noto fonts:

[image]

Notice that only the Noto CJK fonts have this character implemented.  This is a screenshot from the first 12 fonts at https://fonts.google.com/?preview.text=%E2%BF%BB&preview.layout=grid&lang=en_Latn which lists all English fonts:

[image]

The 4 fonts you can't see the name of are Roboto, Kanchenjunga, Tagesschrift, and Coral Pixels, left to right.  As I noted in my PR, I scrolled through the entire list of 1,599 English fonts from my link, and only saw 4 fonts that implemented the character.  Further review of those 4 fonts show they also support at least one CJK language, but they seem to be the only ones outside of Noto's CJK fonts that support this character, even of the fonts that show when selecting a CJK language.

And just to show I have nothing wacky in my Firefox font configuration, here's that screenshot:

[image]

The only thing I've done is set "serif" to mean Noto Sans, and set the monospace font to Noto Sans Mono instead of DejaVu Sans Mono.  I don't have Noto CJK fonts installed, as I don't regularly work with CJK languages.  I have not touched my fontconfig configuration, so it's likely whatever Debian ships.  And if I had a broken configuration (besides missing fonts), why would there be some fonts that do work from Google Fonts?

systemcrash, 2025-05-11:
Your OS / browser config is still breaking something. There is nothing wrong with the fonts. Every single noto font available implements the character.

https://fonts.google.com/noto/fonts?preview.text=%E2%BF%BB&preview.layout=grid 

[image]

https://fonts.google.com/?preview.text=%E2%BF%BB&preview.layout=grid&lang=en_Latn 

[image]

dwfreed, 2025-05-11:
No, that is *your* system substituting fonts when the font that Google wants to use on the page doesn't implement the character.  I downloaded Noto Sans Simplified Chinese, Noto Sans, Inter, Open Sans, and Roboto from Google's site, and fed them to fc-query (which just reads from the font files, not my system, and I've attached the strace file to prove it), and of those 5 fonts, only Noto Sans Simplified Chinese includes U+2FFB in its charset data:

```
$ fc-query -f '%{fullname}: %{charset}\n' Noto_Sans_SC/static/NotoSansSC-Regular.ttf Noto_Sans/static/NotoSans-Regular.ttf Inter/static/Inter_18pt-Regular.ttf Open_Sans/static/OpenSans-Regular.ttf Roboto/static/Roboto-Regular.ttf

Noto Sans SC Regular: 20-7e a0-103 110-113 11a-11b 128-12b 143-144 147-148 14c-14f 152-153 168-16d 192 1a0-1a1 1af-1b0 1cd-1dc 1f8-1f9 251 261 2bb 2c7 2c9-2cb 2d9 2ea-2eb 300-301 304 307 30c 391-3a1 3a3-3a9 3b1-3c9 401 410-44f 451 1e3e-1e3f 1ea0-1ef9 2002-2003 2010-2016 2018-201a 201c-201e 2020-2022 2025-2027 2030 2032-2033 2035 2039-203c 2042 2047-2049 2051 2074 20a9 20ab-20ac 20dd-20de 2100 2103 2105 2109-210a 210f 2113 2116 2121-2122 2126-2127 212b 212e 2135 213b 2160-216b 2170-217b 2190-2199 21b8-21b9 21c4-21c6 21cb-21cc 21d0 21d2 21d4 21e6-21e9 21f5 2200 2202-2203 2205-220b 220f 2211-2213 2215 221a 221d-2220 2223 2225-222e 2234-2237 223d 2243 2245 2248 224c 2260-2262 2264-2267 226a-226b 226e-226f 2272-2273 2276-2277 2282-2287 228a-228b 2295-2299 22a0 22a5 22bf 22da-22db 22ef 2305-2307 2312 2318 2329-232a 23b0-23b1 23be-23cc 23ce 23da-23db 2423 2460-25ab 25b1-25b3 25b6-25b7 25bc-25bd 25c0-25c1 25c6-25c7 25c9-25cc 25ce-25d3 25e2-25e6 25ef 2600-2603 2605-2606 2609 260e-260f 2616-2617 261c-261f 262f 2640-2642 2660-266f 2672-267d 26a0 26bd-26be 2702 2713 271a 273d 273f-2740 2756 2776-2793 27a1 2934-2935 29bf 29fa-29fb 2b05-2b07 2b1a 2b95 2e3a-2e3b 2e80-2e99 2e9b-2ef3 2f00-2fd5 2ff0-2ffb 3000-303f 3041-3096 3099-30ff 3105-312f 3131-3163 3165-318e 3190-31bb 31c0-31e3 31f0-321e 3220-332b 332d-4db5 4e00-9fef f900-f903 f905-f906 f90b-f90c f915 f917-f91a f921 f92c-f92d f92f f931-f935 f937-f93a f943 f947-f94a f94e f952-f953 f95e f962-f967 f96e f972 f974 f976 f979-f97b f97e-f980 f984-f985 f98a-f98c f98e f993 f995 f998 f9ae f9b3 f9b7 f9bb f9bd-f9be f9c0 f9c5-f9c6 f9d0 f9d8-f9d9 f9dc-f9e0 f9e2-f9e4 f9e7 f9e9 f9f1 f9f4-f9f5 f9fa f9fd f9ff fa02 fa05-fa08 fa0a fa0c-fa0f fa11 fa13-fa14 fa18 fa1f-fa21 fa23-fa24 fa27-fa29 fa2f fa33-fa35 fa37-fa38 fa3a fa47 fa49 fa4b fa5d-fa5e fb00-fb04 fe10-fe19 fe30-fe52 fe54-fe66 fe68-fe6b ff01-ff9f ffa1-ffbe ffc2-ffc7 ffca-ffcf ffd2-ffd7 ffda-ffdc ffe0-ffe6 ffe8-ffee 1f100-1f10c 1f110-1f16c 1f170-1f1ac 1f200-1f202 1f210-1f23b 1f240-1f248 1f250-1f251 20087 20089 200cc 20164 20628 20676 20cd0 2139a 21413 215d7 2298f 235cb 23c97-23c98 23e23 241fe 2420e 248e9 249db 24a01 24a7d 24ac9 25532 25562 255a8 25ad7 25ed7 26221 2648d 26676 2677c 26951 26b5c 26c21 278b2 27eaf 27fb7 27ff9 28408 28678 28695 287e0 28b49 28c47 28c4f 28c51 28c54 28e0f 28e99 2967f 29810 29f7e 29f83 29f8c 2a7dd 2a8fb 2a917 2aa30 2aa36 2aa58 2afa2 2b127-2b128 2b137-2b138 2b1ed 2b300 2b363 2b36f 2b372 2b37d 2b404 2b410 2b413 2b461 2b4e7 2b4ef 2b4f6 2b4f9 2b50d-2b50e 2b536 2b5ae-2b5af 2b5b3 2b5e7 2b5f4 2b61c-2b61d 2b626-2b628 2b62a 2b62c 2b695-2b696 2b6ad 2b6ed 2b7a9 2b7c5 2b7e6 2b7f9 2b7fc 2b806 2b80a 2b81c 2b8b8 2bac7 2bb5f 2bb62 2bb7c 2bb83 2bc1b 2bd77 2bd87 2bdf7 2be29 2c029-2c02a 2c0a9 2c0ca 2c1d5 2c1d9 2c1f9 2c27c 2c288 2c2a4 2c317 2c35b 2c361 2c364 2c488 2c494 2c497 2c542 2c613 2c618 2c621 2c629 2c62b-2c62d 2c62f 2c642 2c64a-2c64b 2c72c 2c72f 2c79f 2c7c1 2c7fd 2c8d9 2c8de 2c8e1 2c8f3 2c907 2c90a 2c91d 2ca02 2ca0e 2ca7d 2caa9 2cb29 2cb2d-2cb2e 2cb31 2cb38-2cb39 2cb3b 2cb3f 2cb41 2cb4a 2cb4e 2cb5a-2cb5b 2cb64 2cb69 2cb6c 2cb6f 2cb73 2cb76 2cb78 2cb7c 2cbb1 2cbbf-2cbc0 2cbce 2cc56 2cc5f 2ccf5-2ccf6 2ccfd 2ccff 2cd02-2cd03 2cd0a 2cd8b 2cd8d 2cd8f-2cd90 2cd9f-2cda0 2cda8 2cdad-2cdae 2cdd5 2ce18 2ce1a 2ce23 2ce26 2ce2a 2ce7c 2ce88 2ce93 2d544 2f884 2f8b6 30edd-30ede 3106c

Noto Sans Regular: 20-7e a0-377 37a-37f 384-38a 38c 38e-3a1 3a3-3e1 3f0-52f 900-97f 10fb 1ab0-1ac0 1ac5 1ac7-1ace 1c80-1c88 1d00-1df9 1dfb-1f15 1f18-1f1d 1f20-1f45 1f48-1f4d 1f50-1f57 1f59 1f5b 1f5d 1f5f-1f7d 1f80-1fb4 1fb6-1fc4 1fc6-1fd3 1fd6-1fdb 1fdd-1fef 1ff2-1ff4 1ff6-1ffe 2000-2064 2066-2071 2074-208e 2090-209c 20a0-20c0 20f0 2100-215f 2183-2184 2189 2212 25cc 2c60-2c7f 2de0-2e5d a640-a69f a700-a7ca a7d0-a7d1 a7d3 a7d5-a7d9 a7f2-a7ff a8ff a92e ab30-ab6b fb00-fb06 fe00 fe20-fe2f feff fffc-fffd 10780-10785 10787-107b0 107b2-107ba 1df00-1df1e

Inter 18pt Regular: 20-7e a0-ac ae-148 14a-1c3 1c5-254 256-27b 27e-284 286-290 292-2a4 2a6-304 306-30a 30c 30f 313 315 31b 323 326-328 32c 337-338 342-343 346-36f 374-376 37a-37f 384-38a 38c 38e-3a1 3a3-3d7 3dc-3dd 3f0-3f6 3f9-3fa 3fc-479 480-49d 4a0-4ff 52f e3f 1d00 1d0d 1d1b 1d43 1d47-1d49 1d4d 1d4f-1d50 1d52 1d56-1d58 1d5b 1d62-1d65 1d9c 1da0 1dbb 1dbf-1df5 1dfc-1e9b 1e9d-1f15 1f18-1f1d 1f20-1f45 1f48-1f4d 1f50-1f57 1f59 1f5b 1f5d 1f5f-1f7d 1f80-1fb4 1fb6-1fc4 1fc6-1fd3 1fd6-1fdb 1fdd-1fef 1ff2-1ff4 1ff6-1ffe 2000-200b 2010-2027 202f-2055 2057 205f 2070-2071 2074-208e 2090-209c 20a0-20af 20b1-20b5 20b8-20ba 20bc-20bf 20db-20de 20e8 20f0 2100-2101 2103 2105-2106 2109 2113 2116-2117 211e-2122 2126 212a-212b 212e 2132 213b 214d 2150-217f 2183-2186 2189 2190-2199 21a9-21aa 21b0-21b1 21b3-21b5 21ba-21bb 21d0 21d2 21d4 21de-21df 21e4-21e5 21e7 21ea 2202 2205-2206 220f 2211-2212 221a 221e 222b 2236 2248 2260 2264-2265 2295-2298 2303-2 [... cut ...] et data, so you can actually look at the characters supported.  Once you see a font that doesn't include 2ffb in the charset list, generally none of the fonts after that will either, unless you've done some wild things with your fontconfig configuration.  If you move the files listed that do support U+2FFB out of the font search path, or remove the package(s) owning them, I bet you'll see the same replacement character I do for all fonts on Google Fonts that don't actually support the character.

commit c71dbb61 (2025-06-01): luci-base: remove bad Unicode on clone button

This Unicode character is intended to be followed by 2 more characters
which are combined. It's entirely unnecessary, so just remove it.

Fixes: #7754
closed 2025-06-01 by dibdot