openwrt/luci #7757: LuCI does not show login screen, nothing helpful in logs

schuettecarsten, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

Sone mid of April, LuCI does not show the login screen correctly. The browser redirects from / to /cgi-bin/luci as expected, but then nothing is shown. I can see some HTML code in the response, but it looks like something gets stuck on the server and the UI is never shown.

Using `ubus monitor`, I can see a lot of traffic, no errors, nothing helpful. No output in `dmesg` or `logread`. I have searched for helpful information how to debug LuCI, no idea what's going on.

### Expected behaviour

LuCI login shoud be shown.

### Steps to reproduce

Open LuCI from browser.

### Additional Information

```shell
{
        "kernel": "6.6.88",
        "hostname": "GatewayDummi",
        "system": "ARMv8 Processor rev 3",
        "model": "Raspberry Pi 4 Model B Rev 1.5",
        "board_name": "raspberrypi,4-model-b",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "firmware_url": "https://downloads.openwrt.org/",
                "revision": "r29352+9-0f9af6dcd9",
                "target": "bcm27xx/bcm2711",
                "description": "OpenWrt SNAPSHOT r29352+9-0f9af6dcd9",
                "builddate": "1745959428"
        }
}
```

### What browsers do you see the problem on?

Microsoft Edge

### Relevant log output

```shell

```

jow-, 2025-05-01:
Please run /www/cgi-bin/luci as command on the cli and look fir errors or crashes.

Also check the browser JavaScript/ debug console for errors

schuettecarsten, 2025-05-01:
When I run `/www/cgi-bin/luci`, I get the same HTML output that I can see when I observe the Response details on the browser's Network tab. See below. Also attached output from Browser console.

```
root@GatewayDummi:~# /www/cgi-bin/luci
Status: 403 Forbidden
x-luci-login-required: yes
content-type: text/html; charset=UTF-8
cache-control: no-cache
expires: 0
x-frame-options: SAMEORIGIN
x-xss-protection: 1; mode=block
x-content-type-options: nosniff

<!DOCTYPE html>
<html lang="en" >
        <head>
                <meta charset="utf-8">
                <title>GatewayDummi - LuCI</title>
                        <script>
                                var mediaQuery = window.matchMedia('(prefers-color-scheme: dark)'),
                                    rootElement = document.querySelector(':root'),
                                    setDarkMode = function(match) { rootElement.setAttribute('data-darkmode', match.matches) };

                                mediaQuery.addEventListener('change', setDarkMode);
                                setDarkMode(mediaQuery);
                        </script>
                <meta name="viewport" content="initial-scale=1.0">
                <link rel="stylesheet" href="/luci-static/bootstrap/cascade.css?v=25.113.44500~9180f2a">
                <link rel="stylesheet" media="only screen and (max-device-width: 854px)" href="/luci-static/bootstrap/mobile.css?v=25.113.44500~9180f2a" />
                <link rel="icon" href="/luci-static/bootstrap/logo_48.png" sizes="48x48">
                <link rel="icon" href="/luci-static/bootstrap/logo.svg" sizes="any">
                <script src="/admin/translations/en"></script>
                <script src="/luci-static/resources/cbi.js?v=25.113.44500~9180f2a"></script>
        </head>

        <body class="lang_en " data-page="">
<script src="/luci-static/resources/luci.js?v=25.115.55071~1355a6f-1745959428"></script>
<script>
        L = new LuCI({ "media": "\/luci-static\/bootstrap", "resource": "\/luci-static\/resources", "scriptname": null, "pathinfo": null, "documentroot": null, "requestpath": [ ], "dispatchpath": [ ], "pollinterval": 5, "ubuspath": "\/ubus\/", "sessionid": null, "token": null, "nodespec": { "satisfied": true, "action": { "type": "template", "path": "admin_status\/index" }, "depends": { "acl": [ "luci-mod-status-index" ] }, "order": 1, "title": "Overview" }, "apply_rollback": 90, "apply_holdoff": 4, "apply_timeout": 5, "apply_display": 1.5, "rollback_token": null });
</script>

<section hidden>
        <form method="post" class="cbi-map">
                <div class="cbi-section">
                        <div class="cbi-section-node">
                                <div class="cbi-value">
                                        <label class="cbi-value-title" for="luci_username">Username</label>
                                        <div class="cbi-value-field">
                                                <input name="luci_username" id="luci_username" type="text" autocomplete="username" value="root">
                                        </div>
                                </div>
                                <div class="cbi-value">
                                        <label class="cbi-value-title" for="luci_password">Password</label>
                                        <div class="cbi-value-field">
                                                <input name="luci_password" id="luci_password" type="password" autocomplete="current-password">
                                        </div>
                                </div>
                        </div>
                </div>
        </form>

        <hr>

        <button class="btn cbi-button-positive important">Log in</button>
</section>

<div id="view">
        <div class="spinning">Loading view…</div>
        <script>
                L.require('ui').then(function(ui) {
                        ui.instantiateView('bootstrap.sysauth');
                });
        </script>
</div>

        </body>
</html>

<!-- Lua compatibility mode active: yes -->
```

```
luci/:1 
GET https://192.168.10.1/cgi-bin/luci/ 403 (Forbidden)
luci/:1 
            
            
            GET https://192.168.10.1/cgi-bin/luci/ net::ERR_INCOMPLETE_CHUNKED_ENCODING 403 (Forbidden)
luci/:19 
            
            
            GET https://192.168.10.1/cgi-bin/luci/admin/translations/en net::ERR_INCOMPLETE_CHUNKED_ENCODING 200 (OK)
```

schuettecarsten, 2025-05-01:
@jow- Sorry, accidentially closed this issue, it's still there.

schuettecarsten, 2025-05-01:
@jow- I got this error on Firefox:

[image]

schuettecarsten, 2025-05-01:
I did some more things, the LuCI documentation somewhere says that `curl https://192.168.10.1/cgi-bin/luci/rpc/auth -k --data` following by a json should return a session ID. It does not on my system but results in a 404, a HTML response and an OpenSSL error.

schuettecarsten, 2025-05-02:
@jow- Looks like RPC calls fail, while ubus cli works fine on the device. Any idea how to debug this?

deheisenbug, 2025-05-03:
I may have a pretty similar or at least related issue, which may give some more insight? At least from the `/cgi-bin/luci/` request  failing with and the 403 with no further error/output it feels similar.
I did a fresh install of 24.10.01 some weeks ago on an dap-x1860.
```
# ubus call system board
{
	"kernel": "6.6.86",
	"hostname": "dap-x1860-im18a-00",
	"system": "MediaTek MT7621 ver:1 eco:3",
	"model": "D-Link DAP-X1860 A1",
	"board_name": "dlink,dap-x1860-a1",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "24.10.1",
		"revision": "r28597-0425664679",
		"target": "ramips/mt7621",
		"description": "OpenWrt 24.10.1 r28597-0425664679",
		"builddate": "1744562312"
	}
}
```
 Luci works fine over the default lan interface:
```
# curl -v "http://192.168.1.1/cgi-bin/luci/"
*   Trying 192.168.1.1:80...
* Connected to 192.168.1.1 (192.168.1.1) port 80
* using HTTP/1.x
> GET /cgi-bin/luci/ HTTP/1.1
> Host: 192.168.1.1
> User-Agent: curl/8.13.0
> Accept: */*
>
* Request completely sent off
< HTTP/1.1 403 Forbidden
< Connection: Keep-Alive
< Transfer-Encoding: chunked
< Keep-Alive: timeout=20
< x-luci-login-required: yes
< content-type: text/html; charset=UTF-8
< cache-control: no-cache
< expires: 0
< x-frame-options: SAMEORIGIN
< x-xss-protection: 1; mode=block
< x-content-type-options: nosniff
<
<!DOCTYPE html>
[...]
<!-- Lua compatibility mode active: no -->
* Connection #0 to host 192.168.1.1 left intact
```
corresponding `ubus monitor`
```
-> e456e46e #e456e46e          hello: {}
<- e456e46e #00000000         lookup: {"objpath":"system"}
-> e456e46e #00000000           data: {"objpath":"system","objid":2005103538,"objtype":-1539372080,"signature":{"board":{},"info":{},"reboot":{},"watchdog":{"frequency":5,"timeout":5,"magicclose":7,"stop":7},"signal":{"pid":5,"signum":5},"validate_firmware_image":{"path":3},"sysupgrade":{"path":3,"force":7,"backup":3,"prefix":3,"command":3,"options":2}}}
-> e456e46e #00000000         status: {"status":0}
<- e456e46e #778373b2         invoke: {"objid":2005103538,"method":"board","data":{}}
-> 1095b760 #e456e46e         invoke: {"objid":2005103538,"method":"board","data":{},"user":"root","group":"root"}
<- 1095b760 #e456e46e           data: {"objid":2005103538,"data":{"kernel":"6.6.86","hostname":"dap-x1860-im18a-00","system":"MediaTek MT7621 ver:1 eco:3","model":"D-Link DAP-X1860 A1","board_name":"dlink,dap-x1860-a1","rootfs_type":"squashfs","release":{"distribution":"OpenWrt","version":"24.10.1","revision":"r28597-0425664679","target":"ramips/mt7621","description":"OpenWrt 24.10.1 r28597-0425664679","builddate":"1744562312"}}}
-> e456e46e #778373b2           data: {"objid":2005103538,"data":{"kernel":"6.6.86","hostname":"dap-x1860-im18a-00","system":"MediaTek MT7621 ver:1 eco:3","model":"D-Link DAP-X1860 A1","board_name":"dlink,dap-x1860-a1","rootfs_type":"squashfs","release":{"distribution":"OpenWrt","version":"24.10.1","revision":"r28597-0425664679","target":"ramips/mt7621","description":"OpenWrt 24.10.1 r28597-0425664679","builddate":"1744562312"}}}
<- 1095b760 #e456e46e         status: {"status":0,"objid":2005103538}
-> e456e46e #778373b2         status: {"status":0,"objid":2005103538}
<- e456e46e #00000000         lookup: {"objpath":"session"}
-> e456e46e #00000000           data: {"objpath":"session","objid":1788052494,"objtype":12389323,"signature":{"create":{"timeout":5},"list":{"ubus_rpc_session":3},"grant":{"ubus_rpc_session":3,"scope":3,"objects":1},"revoke":{"ubus_rpc_session":3,"scope":3,"objects":1},"access":{"ubus_rpc_session":3,"scope":3,"object":3,"function":3},"set":{"ubus_rpc_session":3,"values":2},"get":{"ubus_rpc_session":3,"keys":1},"unset":{"ubus_rpc_session":3,"keys":1},"destroy":{"ubus_rpc_session":3},"login":{"username":3,"password":3,"timeout":5}}}
-> e456e46e #00000000         status: {"status":0}
<- e456e46e #6a93840e         invoke: {"objid":1788052494,"method":"get","data":{"ubus_rpc_session":"00000000000000000000000000000000","keys":["rollback"]}}
-> c7f5d12a #e456e46e         invoke: {"objid":1788052494,"method":"get","data":{"ubus_rpc_session":"00000000000000000000000000000000","keys":["rollback"]},"user":"root","group":"root"}
<- c7f5d12a #e456e46e           data: {"objid":1788052494,"data":{"values":{"rollback":{}}}}
-> e456e46e #6a93840e           data: {"objid":1788052494,"data":{"values":{"rollback":{}}}}
<- c7f5d12a #e456e46e         status: {"status":0,"objid":1788052494}
[... 78 lines cut ...]
00000000  47 45 54 20 2f 63 67 69  2d 62 69 6e 2f 6c 75 63   GET /cgi -bin/luc
00000010  69 2f 20 48 54 54 50 2f  31 2e 31 0d 0a 48 6f 73   i/ HTTP/ 1.1..Hos
00000020  74 3a 20 31 30 2e 32 33  2e 31 38 2e 31 30 30 0d   t: 10.23 .18.100.
00000030  0a 55 73 65 72 2d 41 67  65 6e 74 3a 20 63 75 72   .User-Ag ent: cur
00000040  6c 2f 38 2e 31 33 2e 30  0d 0a 41 63 63 65 70 74   l/8.13.0 ..Accept
00000050  3a 20 2a 2f 2a 0d 0a 0d  0a                        : */*... .
    00000000  48 54 54 50 2f 31 2e 31  20 34 30 33 20 46 6f 72   HTTP/1.1  403 For
    00000010  62 69 64 64 65 6e 0d 0a  43 6f 6e 6e 65 63 74 69   bidden.. Connecti
    00000020  6f 6e 3a 20 4b 65 65 70  2d 41 6c 69 76 65 0d 0a   on: Keep -Alive..
    00000030  54 72 61 6e 73 66 65 72  2d 45 6e 63 6f 64 69 6e   Transfer -Encodin
    00000040  67 3a 20 63 68 75 6e 6b  65 64 0d 0a               g: chunk ed..
[... like some 1 or 2? minute timeout here?]
```
wireshark screenshot for packet flow/timing:
[image]

[... cut ...] just loads forever / times out.

[image]

Manually executing `/www/cgi-bin/luci` gives identical output to @schuettecarsten 's earlier response. Besides some version string/name differences there is `` in my case, but no errors.

I was debugging luci some years ago, but this was in the lua days. I could not find any reasonable resource on how to debug this ucode setup further. Any hints?

If it turns out to be something completely different let me know and I will move this into a separate issue.

deheisenbug, 2025-05-03:
I just remembered I have a very similar setup at another place and could not remeber this being a problem earlier. The old setup is also a dap-x1860:
```

systemcrash, 2025-05-05:
Might this be related to 7e6741fbae9b122545e80aad8b9e3a5529c442ce and 852fc23b12d4a701ba724a40b6afd4d40a4ad68d? Those are the only commits which 'touch' anything to do with the luci core. I'm doubtful that it is those, since without them, and the new gcc15, they won't compile. 

Things like this can also happen if jumbo frames are set on the send side or some switch in the middle, and the receive side is not set up for it.

deheisenbug, 2025-05-08:
I think i tracked my  down to an mtu problem with shitty network hardware/drivers. (no jumbo frames tho) So i will hide the now irrelevant comments for now.

schuettecarsten, 2025-05-08:
For me I can say, if I use ImageBuilder and create a image, it works. If I use the same revisions to build my own, the error is back.