openwrt/luci #7759: luci-app-upnp: Opening excessive ports in 24.10

badtlc, 2025-05-01:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

luci-app-upnp is working fine for me using OpenWRT 23.05.5 but when I upgrade to OpenWRT 24.10+, the upnp no longer behaves as expcted.  Instead of opening one extern port per one device UPnP request, it is opening several hundred if not thousands of external ports for a single device/single port request.

### Actual behaviour

Device 1 requests port 1a to be opened
luci-app-upnp opens up a random port that is not 1a and maps it to device 1 port 1a
after a few minutes, hundreds of external ports are opened and mapped to device 1 port 1a

### Expected behaviour

When Device 1 requests port 1a to be opened, Luci-app-upnp should open external port 1a and map it to internal port 1a for Device 1 and nothing else.

### Steps to reproduce

1) install OpenWRT 24.10 or 24.10.1
2) Install luci-app-upnp
3) enable upnp and nat-pmp
4) enable UPnP on network device
5) wait a few minutes and check Luci status overview page to verify multiple ports have been opened/mapped for the single device request

### Additional Information

```shell
Status
System
Hostname	OpenWrt
Model	MikroTik RouterBOARD 750Gr3
Architecture	MediaTek MT7621 ver:1 eco:4
Target Platform	ramips/mt7621
Firmware Version	OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-25.116.54444-149d68e
Kernel Version	5.15.167
```

### Relevant log output

```shell

```

badtlc, 2025-05-08:
Is there a chance this is a minipnp issue?

systemcrash, 2025-05-08:
It's probable the device doing the upnp calls might be doing something incorrectly, but (I think) you can configure upnp to limit resources allocated, if necessary.

Self-Hosting-Group, 2025-11-10:
If any of you would like to test or use the fixed package, please get in touch directly via the linked PR.
https://redirect.github.com/openwrt/packages/pull/24988

closed 2025-12-24 by ldir-EDB0