openwrt/luci #7760: Feature request: luci-proto-wireguard to add wireguard peer names to /etc/hosts

jxs, 2025-05-02:
### What would you like to see in luci?

Hi, and thank you for all the work on openwrt!
Would it be possible/make sense to add the Endpoint Host to `etc/hosts` so that dnsmasq would be able resolve reverse queries for the wireguard addresses?

systemcrash, 2025-05-05:
Can you walk me through an example?

this-username-has-been-taken, 2025-05-06:
Let me guess :)

Let's assume that `Local domain` option is set to `myworld.abc`.
If we have a Wireguard interface with a peer named `warehouse` and the peer endpoint (i.e. IP) set to `10.10.10.123` then FDQN `warehouse.myworld.abc` should be resolved as `10.10.10.123`. Did I guess it right?

What if a peer name contains spaces or other non-compliant symbols? In this case the peer's name should be converted to Punycode and this scenario need additional testing.

systemcrash, 2025-05-06:
Reverse queries require a PTR record. dnsmasq supports these via `--host-record` and `--ptr-record`, so in theory tweaking the proto/init scripts to ask dnsmasq to add a 'forward' record should invoke the creation of a reverse (PTR) record. So you don't even need to touch `/etc/hosts`.

The drawbacks of this approach are that once created, any changes on the wg side might not automatically propagate to the DHCP host entries (which is where the records should be set). The most logical placement of some autoconfig would be in hotplug. The last few efforts to handle automatic interface management for wireguard have stalled. 

In any case, luci isn't the place where this kind of config should be done.

jxs, 2025-05-18:
Thanks for the responses folks!

>Let's assume that Local domain option is set to myworld.abc.
If we have a Wireguard interface with a peer named warehouse and the peer Allowed IP is set to 10.10.10.123 then FQDN warehouse.myworld.abc should be resolved as 10.10.10.123. Did I guess it right?

yeah exactly, I suggested the `Endpoint Host` in the first post but noticed it can't be used for that, same as the interface `name` this would need to be an extra field used specially for that. It did not even need to append to the `Local domain` it'd just need to be a FQDN so that it would be reachable.

> In any case, luci isn't the place where this kind of config should be done.

What if this field and feature was entirely managed by Luci? Thou I can understand this may be to complicated to manage.