openwrt/luci #7766: luci-app-uhttpd: When changing certificate location listen_https options get removed and apply fails

jze2, 2025-05-08:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

When I click on HTTPS Certificate (DER or PEM format) and change the location, the HTTPS listeners fields dissapear and the configuration file is updated without listen_https options which causes failure when applying the changes.

### Expected behaviour

Successful application of certificate location change.

### Steps to reproduce

Click on HTTPS Certificate (DER or PEM format) and select another certificate. Then press Save & apply.

### Additional Information

```shell
Hostname	router
Model	ASUS RT-AX53U
Architecture	MediaTek MT7621 ver:1 eco:3
Target Platform	ramips/mt7621
Firmware Version	OpenWrt 24.10.1 r28597-0425664679 / LuCI (HEAD detached at 2ac26e56) branch 25.103.51521~2ac26e5
Kernel Version	6.6.86

luci-app-uhttpd	25.123.45881~e061a5f
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell

```

jze2, 2025-05-08:
The workaround is to SSH into the box and manually add listen_https options back into /etc/config/uhttpd and reboot.

Ramon00, 2025-07-27:
check if it got fixed with this PR: [https://github.com/openwrt/luci/pull/7829](https://github.com/openwrt/luci/pull/7829)