openwrt/luci #7767: Can't able to login into luci page

ParthPrajapati-lab, 2025-05-09:
### How can we otherwise help?

I have tried to change password
i have tried clear cache of firefox browser
now i don't know what i can do please help
If you need any information i will send you
I have used this os:

/ #
/ # cat /etc/os-release
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0-21bbc6fb"
OPENWRT_BOARD="armvirt/64"
OPENWRT_ARCH="aarch64_generic"
OPENWRT_TAINTS="no-all override"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r0-21bbc6fb"

Just For info:  I have ran image on docker because my system based on ubuntu 22.04(jammy)

ParthPrajapati-lab, 2025-05-13:
Yes Luci Packages are installed 
Issue is login page come and username , password insert but not succeed further means it gives invalid username and password

Just for your info:  I changed password  with passwd and ssh login successfully

getjptothepark, 2025-07-17:
Did you get this resolved, its driving me crazy.