openwrt/luci #7781: Feature request: <summary> luci-app-dnsproxy needs maintainer to become official

Mile-Lile, 2025-05-16:
### What would you like to see in luci?

While I was reading [openwrt wiki dnsproxy](https://openwrt.org/docs/guide-user/services/dns/dot_dnsmasq_dnsproxy) page I founded that this app has unofficial [luci support](https://github.com/muink/luci-app-dnsproxy) that needs maintainer to become official.

@systemcrash can someone look at this? This is the only working DOH3/DOQ client app on Openwrt... it is working very good for me:

[image]

systemcrash, 2025-05-16:
Is this DoT? DoT is dead. DoH seems to be the way everything is going - and we have an app for that. If this works, it doesn't need a maintainer - it's in the repo and it's available :)

Mile-Lile, 2025-05-17:
It is the only package on openwrt that can do h3 and quic dns client... DoQ and DoH3 (even SmartDns still cannot do it...)... it would be nice to be official, this way you need to add insecure repo...

systemcrash, 2025-05-17:
Oh, sorry, I see now. It's an external app. :/ 

Seems simpler to add the fantastic repo as an extra repo.

Mile-Lile, 2025-05-17:
dnsproxy is made by AdGuard team but it is part of official openwrt packages... GUI (luci-app-dnsproxy) is NOT part of Openwrt so if you use openwrt firmware selector you cannot build it with dnsproxy GUI:( downside is that dnsproxy is written in Go (softwarebloat) and has big size 10MB but only client that supports h3 and quic... I liked very much dnsproxy made by aaron but it is limited with new protocols so I switched to this one... thats why I pinged you... btw thank you...

systemcrash, 2025-05-17:
Currently there's no expectation that the luci app is in the openwrt repo :) 

The fantastic repo has https, does it not?
https://fantastic-packages.github.io/releases/

systemcrash, 2025-06-06:
The steps to add a(n external) repo inspired me to add repo key management in acdffc3d1dd8a4a427a2c81d046eb38080127da4, at least as a method to simplify those steps. Give that a try and let me know how it works out for you. I've tested it reasonably well, and it should be able to go into 24 already.

Mile-Lile, 2025-06-06:
thank you... for now I have some kind of error let me try first to upgrade complete firmware

[image]

Mile-Lile, 2025-06-07:
Same after upgrade "dir is not defined"

systemcrash, 2025-06-07:
I must have squashed the wrong commit :/  I've pushed a fix.  If you want to try it earlier:
```diff
function listKeyFiles() {
-	return fs.list(dir).then(entries =>
+	return fs.list(KEYDIR).then(entries =>
```

Mile-Lile, 2025-06-09:
it is working @systemcrash thank you very much

[image]