openwrt/luci #7789: luci-app-pbr: XHR request timed out

Menion2k, 2025-05-28:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

[image]

### Actual behaviour

1. Overview page with "?" and XHR error in Policy Routing page

### Expected behaviour

1. Everything is show properly

### Steps to reproduce

1. Install luci-app-pbr on 24.10.1
2. Upgraded package
3. Login

### Additional Information

```shell
{
        "kernel": "6.6.86",
        "hostname": "MenionPiRouter",
        "system": "ARMv8 Processor rev 3",
        "model": "Raspberry Pi 4 Model B Rev 1.2",
        "board_name": "raspberrypi,4-model-b",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.1",
                "revision": "r28597-0425664679",
                "target": "bcm27xx/bcm2711",
                "description": "OpenWrt 24.10.1 r28597-0425664679",
                "builddate": "1744562312"
        }
}
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell

```

systemcrash, 2025-05-28:
@stangri 

Also, note that this does happen from time to time. One cause is typically a timeout for a long-running b/g operation.

Menion2k, 2025-06-06:
With the latest luci package update in the 24.10.1 feed the problem is solved

closed 2025-06-06 by Menion2k