openwrt/luci #7801: Request to Clarify DHCP Reservation Label in LuCI

nopcarrier, 2025-05-30:
### What would you like to see in luci?

Hello OpenWRT team,

I appreciate the robust features of OpenWRT and its LuCI interface. I have a suggestion to improve clarity in the DHCP Leases section.

Currently, the button labeled "Set Static" is used to create a DHCP reservation for a device, ensuring it receives the same IP address based on its MAC address. However, the term "Set Static" can be misleading, as it traditionally refers to manually configuring a static IP address on the device itself, bypassing DHCP.

To enhance clarity and align with standard networking terminology, I propose renaming this button to one of the following:

"Add DHCP Reservation"

"Create Static Lease"

"Reserve IP Address"

This change would help users better understand the function and prevent potential confusion.

Thank you for considering this suggestion.

Best regards, 
Dan H

systemcrash, 2025-05-31:
Hover text is possible.

closed 2025-06-29 by systemcrash

efahl, 2025-06-29:
@nopcarrier What version OpenWrt do you use?  Fix might need backport for you to see it...