openwrt/luci #7802: luci-compat: ipnet4 and ipnet6 DataTypes in LuCI Reject Valid CIDR Inputs

lazyoop, 2025-06-02:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

When developing LuCI plugins for OpenWrt, I encountered an issue with the `ipnet4` and `ipnet6` datatypes. Specifically, using `o.datatype = "ipnet4"` or `o.datatype = "ipnet6"` to validate user inputs always fails, even when valid network addresses (e.g., 192.168.1.1/24 or 2001:db8::1/64) are provided.

The issue seems to stem from the code in [datatypes.lua](https://github.com/openwrt/luci/blob/d03f135529437d47e126c7333a1992dccb2fc79f/modules/luci-compat/luasrc/cbi/datatypes.lua#L166-L176):
```lua
function ipnet4(val)
	local ip, mask = val:match("^([^/]+)/([^/]+)$")

	return ip4addr(ip) and ip4addr(mask)
end

function ipnet6(val)
	local ip, mask = val:match("^([^/]+)/([^/]+)$")

	return ip6addr(ip) and ip6addr(mask)
end
```

Here, `ipnet4` incorrectly passes the CIDR mask (such as 24) to ip4addr(mask), which fails since "24" is not a valid IPv4 address. The same issue applies to `ipnet6`, which misinterprets the IPv6 mask.

### Expected behaviour

When the user provides a valid network address, such as 192.168.1.1/24 or 2001:db8::1/64, it should be recognized as a valid network address by `ipnet4` or `ipnet6` validation.

### Steps to reproduce

1. Add the following code to a LuCI CBI form: 
```lua
o = s:option(Value, "test_network", "Test Network")
o.datatype = "ipnet4"  -- or "ipnet6"
 ```
2. In the web UI, enter 192.168.1.1/24 or 2001:db8::1/64.
3. The form validation fails, incorrectly rejecting the valid input.

### Additional Information

```shell
I have tried both the LuCI `21.02 branch` and the latest `master branch`, and the issue exists in both.
I have not checked other LuCI branches.
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell

```

jow-, 2025-06-02:
The ipnet validators are supposed to only accept ipaddr/netmask notation. If your input expects cidr, use the cidr validators. If it accepts both cidr and addr/mask notation, use `or(cidr4, ipnet4)`, if it accepts all of cidr, addr/mask or bare addr notation, use `ipmask4` (or `or(cidr4, ipnet4, ip4addr)`)

lazyoop, 2025-06-03:
Thank you, I understand.