openwrt/luci #7804: Numerical Etc/GMT timezone variants are reversed

RTechSn, 2025-06-02:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

When selecting Etc/GMT+, your timezone will be set to Etc/GMT-.
When selecting Etc/GMT-, your timezone will be set to Etc/GMT+.

### Expected behaviour

Correct timezone setting.

### Steps to reproduce

1. Open Luci, go to System -> System,
2. set your timezone to Etc/GMT+something or Etc/GMT-something.

### Additional Information

```shell
"distribution": "OpenWrt",
"version": "24.10.1",
"revision": "r28597-0425664679",
"target": "x86/64",
"description": "OpenWrt 24.10.1 r28597-0425664679",
"builddate": "1744562312"
```

### Relevant log output

```shell

```

closed 2025-06-10 by jow-

RTechSn, 2025-06-11:
Just curious how should a user know that a POSIX standard is used there. Named timezones use the opposite operation, shouldn't elements of the same list follow the same standard?

hnyman, 2025-06-11:
You should dicsuss that with the upstream,  IANA's time zone list maintainers, not us here... 
We are just taking the global definitions for both the timezone names and the definition strings.