openwrt/luci #7806: Why is the remove function of an option not called if the whole section is removed

feckert, 2025-06-03:
I'm a bit confused right now.
If I create a new section and overwrite the `o.write` function of an option so that it creates and changes other options in other uci files, then this is working as expected.

But I want also remove this other uci options that I have previous added, if I remove the whole section. But this does not work. The `o.remove`  function is not being called on section remove?

I would have expected that all `o.remove` function get called if the whole section gets removed.

Is this a bug or is this intended?

Example code:
```
'use strict';
'require form';
'require view';
'require uci';
reeturn view.extend({
	render: function() {

		let m, s, o;

		m = new form.Map('template', _('Template'));

		s = m.section(form.GridSection, 'template');
		s.addremove = true;
		s.anonymous = false;
		s.uciconfig = "template";

		o = s.option(form.ListValue, 'name', _('Name'));
		o.value('ftth_pppoe_v4', _('FTTH PPPOE IPv4'));
		o.value('ftth_static_v4', _('FTTH Static IPv4'));
		o.value('ftth_dhcp_v4', _('FTTH DHCP IPv4'));
		o.value('ftth_pppoe_v4v6', _('FTTH PPPOE IPv4/IPv6'));
		o.value('ftth_static_v4v6', _('FTTH Static IPv4/IPv6'));
		o.value('ftth_dhcp_v4v6', _('FTTH DHCP IPv4/IPv6'));
		o.write = function(section_id, value) {
			uci.add('network', 'interface', section_id);
			uci.set('network', section_id, 'device', 'sfp');

			console.log("Add section_id: '%s' value: '%s'", section_id, value);
		};
		o.remove = function(section_id) {
			console.log("Remove section_id: '%s'", section_id);
		};

                o = s.option(form.Value, 'username', _('Username'));
                o.rmempty = false;
                o.depends('name', 'ftth_pppoe_v4');
                o.depends('name', 'ftth_pppoe_v4v6');

                o = s.option(form.Value, 'password', _('Password'));
                o.password = true;
                o.rmempty = false;
                o.depends('name', 'ftth_pppoe_v4');
                o.depends('name', 'ftth_pppoe_v4v6');

                o = s.option(form.Value, 'ipaddr', _('IPv4 address'));
                o.datatype = 'ip4addr("nomask")';
                o.depends('name', 'ftth_static_v4');
                o.depends('name', 'ftth_static_v4v6');

                o = s.option(form.Value, 'netmask', _('Netmask'));
                o.datatype = 'ip4addr';
                o.depends('name', 'ftth_static_v4');
                o.depends('name', 'ftth_static_v4v6');

                o = s.option(form.Flag, 'auto', _('Bring up on boot'));
                o.depends('name', 'ftth_pppoe_v4');
                o.depends('name', 'ftth_static_v4');
                o.depends('name', 'ftth_dhcp_v4');
                o.depends('name', 'ftth_pppoe_v4v6');
                o.depends('name', 'ftth_static_v4v6');
                o.depends('name', 'ftth_dhcp_v4v6');

                return m.render();
        },
});

feckert, 2025-06-03:
Thank you for your feedback.
Then that's how it's meant to be.
Then I guess I have no choice and have to overwrite handlRemove for this section to do the magic there.

systemcrash, 2025-06-03:
> overwrite handleRemove for this section 

that's the way I would recommend.