openwrt/luci #7819: Cannot input custom value on mobile

ledlamp, 2025-06-15:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

when I try to input custom value on my android phone, nothing happens when I press the enter/return key on the touch keyboard

### Expected behaviour

pressing enter/return is the only way to use the custom value but it's not working on mobile

### Steps to reproduce

go to a value input with a dropdown of suggested values like the one in the screenshot and try to input custom value

### Additional Information

```shell
OpenWrt 24.10.1 r28597-0425664679 / LuCI openwrt-24.10 branch 25.161.59576~cb94dcc
```

### What browsers do you see the problem on?

Chrome

### Relevant log output

```shell

```

ledlamp, 2025-06-15:
It works on Firefox, it doesn't work in Chrome or Vivaldi

Ramon00, 2025-07-25:
Have a look at the PR I just created [https://github.com/openwrt/luci/pull/7876](https://github.com/openwrt/luci/pull/7876). This should fix the issue. Please test.

closed 2025-07-28 by systemcrash