openwrt/luci #7865: Feature request: nlbwmon to store the information even after rebooting the router

learner8094, 2025-07-16:
### What would you like to see in luci?

when i reboot the router luci-app-nlbwmon reset the couting, i hope that you add the option to store the information even after restarting the router.

learner8094, 2025-07-29:
> Add persistent storage to nlbwmon so data isn’t lost after router reboot

can you explain how to do this?
 the database directory in the app settings is set to: "/var/lib/nlbwmon" and i think that this is not a temporary folder?

hnyman, 2025-07-30:
/var is symlinked to /tmp in the default config, so it is ramdisk.

OpenWrt is really conscious about flash wear, so there are several things like this, where the practice deviates from POSIX.

hnyman, 2025-07-30:
If you want to move the database to a directory on permanent flash, you can use pretty much any directory. E.g. create /etc/nlbwmon

But make sure that you do adjust nlbwmon options to not write into flash at every 30 seconds or so. To minimise flash wear, the writing of  should be done maybe 1-2 times per day. See the "Commit interval" option.

closed 2025-07-30 by hnyman