openwrt/luci #7878: luci-app-upnp: No active port forwarding listed

yangfl, 2025-07-31:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

Go to http://xx.xx.xx.xx/cgi-bin/luci/admin/services/upnp

[image]

No active port forwarding listed even there are port forwardings.

### Expected behaviour

[image]

### Steps to reproduce

1. Make sure there is a port forwarding. 

```
(pc) $ upnpc -r 8888 tcp
...
external xx.xx.xx.xx:8888 TCP is redirected to internal xx.xx.xx.xx:0 (duration=604800)
```

```
# cat /var/run/miniupnpd.leases
TCP:8888:xx.xx.xx.xx:8888:1754581151:libminiupnpc
# nft --handle list chain inet fw4 upnp_prerouting 2>/dev/null
table inet fw4 {
	chain upnp_prerouting { # handle 41
		iif "pppoe-wan" @nh,72,8 0x6 th dport 8888 dnat ip to xx.xx.xx.xx:8888 # handle 3783
	}
}
# /etc/init.d/uhttpd restart
# /etc/init.d/rpcd restart
```

2. Go to http://xx.xx.xx.xx/cgi-bin/luci/admin/services/upnp , no port forwarding listed.

3. Replace

https://github.com/openwrt/luci/blob/6da5e65981734021932b8d339ce7a386eb8f8476/applications/luci-app-upnp/root/usr/share/rpcd/ucode/luci.upnp#L89

with `return { rules };`, then
```
# /etc/init.d/uhttpd restart
# /etc/init.d/rpcd restart
```

Shown as Pic. 2.

### Additional Information

```shell
OpenWrt 24.10.2 r28739-d9340319c6 / LuCI openwrt-24.10 branch 25.206.68186~6da5e65
```

### Relevant log output

```shell

```

nbd168, 2025-08-05:
Did you upgrade all ucode related packages (including libucode), or just ucode-mod-ubus? The error should only appear when mixing packages from different versions on the same system.

systemcrash, 2025-08-09:
> Could the LuCI plugin be changed (e.g. by using a different call than `ubus.defer`) to find a solution that doesn't require an rpcd fix?

I haven't tested it but read the initial post. 

This bug will persist until there's an effort to merge jows proposed or a synonymous fix and bump rpcd in master, and pick that to 24.

Sam-arch99, 2025-08-10:
Same issue with a freshly created 24.10.2 image from yesterday. It was working fine with my initial 24.10.2 build from weeks ago.
I hope that it get fixed soon with 24.10.*

Sam-arch99, 2025-08-11:
> > Is this valid for OpenWrt 24.10.2? opkg update doesn't show any packages that can be upgraded in my case and my last build is from 2 days ago.
> 
> To find out if the necessary package update has been built for the required architecture (24.10.x): The fix is included in `rpcd` in the `base` directory if the package is created after 10 august (or late). https://mirror-03.infra.openwrt.org/releases/24.10.2/packages/

No luck for me with a GL-MT6000.

Package: rpcd
Version: 2024.09.17~9f4b86e7-r1
Depends: libc, libubus20250102, libubox20240329, libuci20250120, libblobmsg-json20240329, libjson-c5
License: ISC
Section: utils
Architecture: aarch64_cortex-a53
Installed-Size: 81920

It's the same version that is currently in my 24.10.2 build from 2 days ago. It's kinda confusing that I didn't had this issue with my initial 24.10.2 build from weeks ago but I guess I can't do anything else than wait.

sc16me, 2025-08-13:
/usr/share/rpcd/ucode/luci.upnp
```
// Copyright 2022 Jo-Philipp Wich <jo@mein.io>
// Licensed to the public under the Apache License 2.0.

'use strict';

import { access, open, popen } from 'fs';
import { connect } from 'ubus';
import { cursor } from 'uci';

// Establish ubus connection persistently outside of the call handler scope to
// prevent premature GC'ing. Can be moved into `get_status` callback once
// https://github.com/jow-/ucode/commit/a58fe4709f661b5f28e26701ea8638efccf5aeb6
// is merged.
const ubus = connect();

const methods = {
	get_status: {
		call: function(req) {
			const uci = cursor();

			const rules = [];
			const leases = [];

			const leasefile = open(uci.get('upnpd', 'config', 'upnp_lease_file'), 'r');

			if (leasefile) {
				for (let line = leasefile.read('line'); length(line); line = leasefile.read('line')) {
					const record = split(line, ':', 6);

					if (length(record) == 6) {
						push(leases, {
							proto: uc(record[0]),
							extport: +record[1],
							intaddr: arrtoip(iptoarr(record[2])),
							intport: +record[3],
							expires: record[4] - timelocal(localtime()),
							description: trim(record[5])
						});
					}
				}

				leasefile.close();
			}

			const ipt = popen('iptables --line-numbers -t nat -xnvL MINIUPNPD 2>/dev/null');

			if (ipt) {
				for (let line = ipt.read('line'); length(line); line = ipt.read('line')) {
					let m = match(line, /^([0-9]+)\s+([a-z]+).+dpt:([0-9]+) to:(\S+):([0-9]+)/);

					if (m) {
						push(rules, {
							num: m[1],
							proto: uc(m[2]),
							extport: +m[3],
							intaddr: arrtoip(iptoarr(m[4])),
							intport: +m[5],
							descr: ''
						});
					}
				}

				ipt.close();
			}

			const nft = popen('nft --handle list chain inet fw4 upnp_prerouting 2>/dev/null');

			if (nft) {
				for (let line = nft.read('line'), num = 1; length(line); line = nft.read('line')) {
					let m = match(line, /^\t\tiif ".+" @nh,72,8 (0x6|0x11) th dport ([0-9]+) dnat ip to ([0-9.]+):([0-9]+)/);

					if (m) {
						push(rules, {
							num: `${num}`,
							proto: (m[1] == '0x6') ? 'TCP' : 'UDP',
							extport: +m[2],
							intaddr: arrtoip(iptoarr(m[3])),
							intport: +m[4],
							descr: ''
						});

						num++;
					}
				}

				nft.close();
			}

			for (let rule in rules) {
				for (let lease in leases) {
					if (lease.proto == rule.proto &&
					    lease.intaddr == rule.intaddr &&
					    lease.intport == rule.intport &&
					    lease.extport == rule.extport)
					{
						rule.descr = lease.description;
						rule.expires = lease.expires;
						break;
					}
				}
			}

			const dhcp_leases = open('/var/dhcp.leases', 'r');
			const host_hints = {};

			if (dhcp_leases) {
				for (let line = dhcp_leases.read('line'); length(line); line = dhcp_leases.read('line')) {
					const parts = split(line, ' ', 5);
					if (length(parts) >= 4) {
						const ip = parts[2];
						const hostname = parts[3];
						if (ip && hostname && hostname != '*') {
							host_hints[ip] = hostname;
						}
					}
				}
				dhcp_leases.close();
			}

			for (let rule in rules) {
				if (rule.intaddr in host_hints) {
					rule.host_hint = host_hints[rule.intaddr];
				}
			}

			req.reply({ rules });
		}
	},

	delete_rule: {
		args: { token: 'token' },
		call: function(req) {
			const idx = +req.args?.token;

			if (idx > 0) {
				const uci = cursor();
				const leasefile = uci.get('upnpd', 'config', 'upnp_lease_file');

				if (access(leasefile)) {
					system(['sed', '-i', '-e', `${idx}d`, leasefile]);
					system(['/etc/init.d/miniupnpd', 'restart']);
				}

				return { result: 'OK' };
			}

			return { result: 'Bad request' };
		}
	}
};

return { 'luci.upnp': methods };

```
This is a temporary and probably not the most beautiful solution to the problem.

Sam-arch99, 2025-08-17:
> /usr/share/rpcd/ucode/luci.upnp
>
> ```
> // Copyright 2022 Jo-Philipp Wich <jo@mein.io>
> // Licensed to the public under the Apache License 2.0.
>
> 'use strict';
>
> import { access, open, popen } from 'fs'; [...]

I just hope that the responsible packages will be updated. I'm on 24.10.2 and no there are no updated packages available yet... :(

timur-davletshin, 2025-08-17:
`rpcd - 2024.09.17~9f4b86e7-r1` (August 12) on 24.10.2 (aarch64_cortex-a53) - problem still exists. No forwards listed.

Sam-arch99, 2025-08-17:
> [@Sam-arch99](https://github.com/Sam-arch99) Therefore, I think your very long comment could be deleted.

I'll just installed a fresh 24.10.2 build with all my packages and the issue is indeed fixed. My bad, I'll delete my comments now!

timur-davletshin, 2025-08-21:
Ucode packages versioned 2025.07.18~3f64c808-r1 is the culprit. After upgrading those it stops reporting ports forwarded by miniupnpd.

timur-davletshin, 2025-08-30:
@Ralith do not upgrade ucode packages to 2025.07.18~3f64c808-r1 just as I said above.

Ralith, 2025-08-30:
> do not upgrade ucode packages to 2025.07.18~3f64c808-r1

Bit late for that. I asked what version contains the fix.

> what's the architecture of your router?

```
cpu model		: MIPS 74Kc V5.0
```

timur-davletshin, 2025-09-01:
@KawaiianPizza your problem is of different nature. All others had it working until recent update.

timur-davletshin, 2025-09-04:
Now it works fine for me after recent upgrade. I can't confirm that some information is missing...
[image]

youryazhuo, 2025-09-05:
had restarted device,but not worked.
ubus call luci.upnp get_status
`{
                        "num": "1",
                        "proto": "TCP",
                        "extport": 19801,
                        "intaddr": "192.168.1.97",
                        "intport": 40292,
                        "descr": ""
                },`
I don't know if it is this related to /usr/share/rpcd/ucode/luci.upnp file code