openwrt/luci #7884: Unbound: got notification with odhcp

hingbong, 2025-08-02:
HI,

https://github.com/openwrt/luci/blame/8d367398d626436e67478da447ce4a4ffe7992e2/applications/luci-app-unbound/htdocs/luci-static/resources/view/unbound/configure.js#L40C3-L40C3

I got notification about odhcp `Note: local DNS is configured to look at odhpcd, but odhpcd UCI lease trigger is incorrectly set:dhcp.odhcpd.leasetrigger='/usr/lib/unbound/odhcpd.sh'`, refrence to link above. May I have an ask, any issues there?

dibdot, 2025-09-18:
@systemcrash Are you planning to fix this wrong notification? According to the unbound readme this is the correct leasetrigger value. To fix this just replace "===" with "!==" in the leasetrig check.

systemcrash, 2025-09-18:
Sure. Oversight during js conversion. 

Closed by f5d7eaa94e3312ea458926901ff5a13930e100dc

closed 2025-09-18 by systemcrash

dibdot, 2025-09-19:
Many thanks!