openwrt/luci #7897: dnsmasq: Default value in LuCI for cache size is wrong, should be 150

schuettecarsten, 2025-08-10:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

Shows 1000 in LuCI, but default value is 150 according to dnsmasq man page and OpenWRT docs:

[image]

Docs:
https://openwrt.org/docs/guide-user/base-system/dhcp

### Actual behaviour

Default value sowas as 1000

### Expected behaviour

Default value should be 150

### Steps to reproduce

Network -> DHCP/DNS -> Limits

### Additional Information

```shell
{
        "kernel": "6.12.40",
        "hostname": "GatewayDummi",
        "system": "ARMv8 Processor rev 3",
        "model": "Raspberry Pi 4 Model B Rev 1.5",
        "board_name": "raspberrypi,4-model-b",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "firmware_url": "https://downloads.openwrt.org/",
                "revision": "r30729-f650f2975c",
                "target": "bcm27xx/bcm2711",
                "description": "OpenWrt SNAPSHOT r30729-f650f2975c",
                "builddate": "1754724347"
        }
}
```

### Relevant log output

```shell

```

stangri, 2025-08-28:
IMHo it might be beneficial to update the default values for append_parm generally and specifically for https://github.com/openwrt/openwrt/blob/08cc7e881e88237aab663472ca0e71ebc1ee3dcb/package/network/services/dnsmasq/files/dnsmasq.init#L1026.

systemcrash, 2025-09-02:
Closed by a2d0f5e2b3fdaca37cc088e40f4cfcb67b23cffe / d968d20a7d89e5afa3d821aa9426faae46cf51cf (24.10)

closed 2025-09-02 by systemcrash