openwrt/luci #7912: luci-app-frpc: Change format to fit FRP v2 version

PunkFleet, 2025-08-21:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

```
Thu Aug 21 13:59:45 2025 daemon.info frpc[7716]: 2025/08/21 05:59:45 [W] [service.go:133] login to server failed: EOF
Thu Aug 21 13:59:45 2025 daemon.info frpc[7716]: 2025/08/21 05:59:45 [I] [root.go:236] frpc service for config file [/var/etc/frpc.ini] stopped
Thu Aug 21 13:59:50 2025 daemon.info frpc[7746]: 2025/08/21 05:59:50 [I] [root.go:220] start frpc service for config file [/var/etc/frpc.ini]
Thu Aug 21 13:59:50 2025 daemon.info frpc[7746]: 2025/08/21 05:59:50 [W] [service.go:133] login to server failed: EOF
Thu Aug 21 13:59:50 2025 daemon.info frpc[7746]: 2025/08/21 05:59:50 [I] [root.go:236] frpc service for config file [/var/etc/frpc.ini] stopped
Thu Aug 21 13:59:55 2025 daemon.info frpc[7792]: 2025/08/21 05:59:55 [I] [root.go:220] start frpc service for config file [/var/etc/frpc.ini]
Thu Aug 21 13:59:57 2025 daemon.info frpc[7792]: 2025/08/21 05:59:57 [W] [service.go:133] login to server failed: EOF
Thu Aug 21 13:59:57 2025 daemon.info frpc[7792]: 2025/08/21 05:59:57 [I] [root.go:236] frpc service for config file [/var/etc/frpc.ini] stopped
Thu Aug 21 14:00:02 2025 daemon.info frpc[7797]: 2025/08/21 06:00:02 [I] [root.go:220] start frpc service for config file [/var/etc/frpc.ini]
Thu Aug 21 14:00:02 2025 daemon.info frpc[7797]: 2025/08/21 06:00:02 [W] [service.go:133] login to server failed: EOF
Thu Aug 21 14:00:02 2025 daemon.info frpc[7797]: 2025/08/21 06:00:02 [I] [root.go:236] frpc service for config file [/var/etc/frpc.ini] stopped
Thu Aug 21 14:00:07 2025 daemon.info frpc[7804]: 2025/08/21 06:00:07 [I] [root.go:220] start frpc service for config file [/var/etc/frpc.ini]
Thu Aug 21 14:00:08 2025 daemon.info frpc[7804]: 2025/08/21 06:00:08 [W] [service.go:133] login to server failed: EOF
Thu Aug 21 14:00:08 2025 daemon.info frpc[7804]: 2025/08/21 06:00:08 [I] [root.go:236] frpc service for config file [/var/etc/frpc.ini] stopped
Thu Aug 21 14:00:13 2025 daemon.info frpc[7829]: 2025/08/21 06:00:13 [I] [root.go:220] start frpc service for config file [/var/etc/frpc.ini]
Thu Aug 21 14:00:14 2025 daemon.info frpc[7829]: 2025/08/21 06:00:14 [W] [service.go:133] login to server failed: EOF
Thu Aug 21 14:00:14 2025 daemon.info frpc[7829]: 2025/08/21 06:00:14 [I] [root.go:236] frpc service for config file [/var/etc/frpc.ini] stopped
Thu Aug 21 14:00:14 2025 daemon.info procd: Instance frpc::instance1 s in a crash loop 6 crashes, 1 seconds since last crash
```

### Actual behaviour

in `/etc/config/frpc`

```
        option token '12345678'
```

### Expected behaviour

in `/etc/config/frpc`

```
        option auth_method 'token'
        option auth_token '12345678'
```

### Steps to reproduce

The existing luci-app-frpc syntax is not compatible with frps v2, resulting in an EOF error.

### Additional Information

```shell
{
        "kernel": "6.12.15-production+truenas",
        "hostname": "Router",
        "system": "12th Gen Intel(R) Core(TM) i5-12600K",
        "model": "Maxsun MS-Challenger B760ITX D5",
        "board_name": "maxsun-ms-challenger-b760itx-d5",
        "rootfs_type": "zfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.2",
                "revision": "r28739-d9340319c6",
                "target": "x86/64",
                "description": "OpenWrt 24.10.2 r28739-d9340319c6",
                "builddate": "1750711236"
        }
}
```

### Relevant log output

```shell

```

systemcrash, 2025-08-22:
@ysc3839

ysc3839, 2025-08-22:
frp is using a new config format, the old INI format is deprecated.
The new config format is using TOML and all option names are changed.
A complete rewrite is required to support new config format.

luci-app-frpc is still using the old INI config format.
According to legacy config example:
https://github.com/fatedier/frp/blob/14253afe2f5deb66cddbd9a0aac37dd31ace7d29/conf/legacy/frpc_legacy_full.ini#L43-L46
`authentication_method` option can be omitted.
I think this is not root cause of this issue.

I have replace frp with WireGuard+firewall forwarding some years ago.
So I can't test if it working now.
I will remove remove myself as maintainer.
Thank you.

laipeng668, 2025-10-21:
The latest frps remains compatible with `ini` configuration files, and OpenWrt continues to use `ini` configuration files. The `option token '12345678'` configuration is normal; please check your other configurations.
https://github.com/fatedier/frp/blob/dev/conf/legacy/frpc_legacy_full.ini

systemcrash, 2025-10-21:
@laipeng668 please see #8024

laipeng668, 2025-10-21:
> [@laipeng668](https://github.com/laipeng668) please see [#8024](https://github.com/openwrt/luci/issues/8024)

Yeah, I am working on it!