openwrt/luci #7947: luci-app-ddns: DDns Service - Log File Viewer cannot be scrolled

sppmasterspp, 2025-09-17:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

When DDNS Log File Viewer is opened the information cannot be viewed because the modal window cannot be scrolled down.

### Expected behaviour

I think there should be a scroll bar so the log can be scrolled down to see the info there.

### Steps to reproduce

Go to services -> Dynamic DNS -> Open a service -> Open the last tab and press READ / REREAD LOG FILE

### Additional Information

```shell
"kernel": "6.12.45",
        "hostname": "Asus",
        "system": "ARMv8 Processor rev 4",
        "model": "Asus RT-AX89X",
        "board_name": "asus,rt-ax89x",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "firmware_url": "https://downloads.openwrt.org/",
                "revision": "r31168+1-3cc85f9d8f",
                "target": "qualcommax/ipq807x",
                "description": "OpenWrt SNAPSHOT r31168+1-3cc85f9d8f",
                "builddate": "1757441291"
```

### What browsers do you see the problem on?

Chrome, Firefox, Microsoft Edge

### Relevant log output

```shell

```

systemcrash, 2025-09-18:
Scrolls fine for me

sppmasterspp, 2025-09-18:
> Scrolls fine for me

Do you use Bootstrap Theme?
I've just tried it and it works with Bootstrap Theme. 
I mostly use Material Theme and it doesn't work with it.

systemcrash, 2025-09-18:
You should be able to click into the window and use your arrow keys with or without control (or alt) to scroll around.

commit 755061bc (2025-09-18): themes: remove overflow-y property; it inhibits readonly view scrolling

Closes #7947
closed 2025-09-18 by systemcrash